pothosware/SoapySDR #29: RTL-SDR Direct Sampling mode?

cjcliffe, 2015-09-14:
I wasn't able to find a way to set the RTL-SDR direct sampling mode via the device controls; is there something I've missed that can be used to toggle this setting?

Thanks!

guruofquality, 2015-09-14:
The RTL is supported the the SoapyOsmo wrapper: https://github.com/pothosware/SoapyOsmo/wiki
So looking at the code for rtl from gr-osmosdr, you should be able to pass the following key/value pair with the device's factory args: ```direct_samp=1```

```
if (dict.count("rtl_xtal"))
    rtl_freq = (unsigned int)boost::lexical_cast< double >( dict["rtl_xtal"] );

  if (dict.count("tuner_xtal"))
    tuner_freq = (unsigned int)boost::lexical_cast< double >( dict["tuner_xtal"] );

  if (dict.count("direct_samp"))
    direct_samp = boost::lexical_cast< unsigned int >( dict["direct_samp"] );

  if (dict.count("offset_tune"))
    offset_tune = boost::lexical_cast< unsigned int >( dict["offset_tune"] );

  _buf_num = _buf_len = _buf_head = _buf_used = _buf_offset = 0;

  if (dict.count("buffers"))
    _buf_num = boost::lexical_cast< unsigned int >( dict["buffers"] );

  if (dict.count("buflen"))
    _buf_len = boost::lexical_cast< unsigned int >( dict["buflen"] );
```

Let me know if that helps.

cjcliffe, 2015-09-14:
Yup that works perfectly -- that covers several settings I was looking for.

Thanks!

closed 2015-09-14 by cjcliffe