pothosware/SoapySDR #30: SoapySDR::Device::make() with multiple RTL-SDR devices always uses first one?

cjcliffe, 2015-09-14:
If I enumerate the RTL devices returned from std::vector<SoapySDR::Kwargs> results = SoapySDR::Device::enumerate(); and attempt to SoapySDR::Device::make(results[x]); a specific result I always get the first RTL device.  

Is there something I might need to add or am I using the results parameters incorrectly?

Similarly I can't get SoapySDRUtil to --probe a specific device.  It looks like the format should be --probe="driver=rtl,rtl=x" where 'x' is the device index according to the --find of:

```
Found device 0
  driver = rtl
  label = Realtek RTL2838UHIDIR SN: 00000003
  rtl = 0
```

where the 'rtl = 0' increases with each device.

guruofquality, 2015-09-14:
That's the right idea. I think that the discovery function in SoapyOsmo is probably not parsing the additional keyword arguments and filtering out the result. I will take a look.

guruofquality, 2015-09-14:
This commit on SoapyOsmo should fix the issue by implementing the device number filtering: https://github.com/pothosware/SoapyOsmo/commit/155d351d7c70b4fcbc7e4eb3680fb470414bcdc4

cjcliffe, 2015-09-14:
Thanks for your quick response -- just tried it with a handful of RTL-Dongles connected and appears to be working correctly now.

closed 2015-09-14 by cjcliffe