pothosware/SoapySDR #31: Enhancement request: SoapySDRPlay module

cjcliffe, 2015-09-14:
SDRPlay is on the list of devices I'm hoping to support for my application through SoapySDR -- though
I'm informed that the OsmoSDR implementation for SDRPlay device is buggy (looks like it's only a few months old) and I was hoping you might be able to investigate adding a native module for the device.

I unfortunately don't have the hardware but @Toontje does and is eagerly standing by for testing anything related to SDRPlay if needed.

Thanks!

guruofquality, 2015-09-14:
Buggy or not, FWIW I can update the gr-osmosdr copy in SoapyOsmo to add the SDRPlay support.

But I dont mind launching another support module just for SDR Play. Thats probably for the best. I looked at the sdr_source_c.cc, its not very much at all.

If I create the github repo, and put together build stubs and stub functions to fill in based on mirsdrapi-rsp.h, can someone complete and test the module? @Toontje ?

And when its all said and done I can throw in the new project w/ the Pothos SDR windows installer: https://github.com/pothosware/PothosSDR/wiki

cjcliffe, 2015-09-15:
That would be great; if you can get a module together I can definitely work with @Toontje on testing builds.

guruofquality, 2015-09-15:
Repo is up: https://github.com/pothosware/SoapySDRPlay

* The repo has a working build system, and stubs for a Soapy SDR device. I tried to fill in some of the functions and left comments about what I think each function should do.
* You should have R/W access to the SoapySDRPlay repo now
* If you have to print something in the driver, use SoapySDR::log or logf, that way we can forward the log messages when using SoapyRemote or Pothos SDR blocks for example: https://pothosware.github.io/SoapySDR/doxygen/latest/Logger_8hpp.html
* I don't know if mutexes are needed, but I suggest enabling c++11 and using std::mutex rather than dragging in all of boost
* So, let me know if you need clarification for how a particular function should be implemented.

Toontje, 2015-09-15:
Yep, i can do the testing of the module. No problem.

Ton.

cjcliffe, 2015-09-15:
@guruofquality excellent -- thanks for this; I'll get the SDRPlay libs compiling and start filling out the functions later this evening and pass on to @Toontje for testing when ready.

closed 2015-10-04 by cjcliffe