pothosware/SoapySDR #40: Debug messages

Toontje, 2015-10-20:
Can we get an option to limit the SoapySDR debug messages?
[image]

guruofquality, 2015-10-20:
100% of the log messages are making it to the display window. There's a few way to tackle this. This reply is sort of a thought dump at the moment:

SoapySDR Logger
===============

* https://github.com/pothosware/SoapySDR/blob/master/lib/LoggerC.cpp

SoapySDR has no log filtering, by default everything goes to the log handler. I think adding an API call to set the log level and an environment variable to also modify the default log level would be a good thing to have.

Pothos SDR log forwarding
================

* https://github.com/pothosware/pothos-sdr/blob/master/Logger.cpp

At this level, a new log handler is installed to forward the logs into the Poco logging subsystem. This is also how the log messages get around when the blocks are in a different process or host machine.

There is another option in here to implement the same kind of log filtering as mentioned above. In addition, the SDR blocks could have a log level themselves so it would be available in the GUI as a block parameter.

Its better to filter out logs earlier than later. Either this level or in SoapySDR, because after that, and there is unnecessary traffic going into the Poco logging subsystem.

Pothos GUI message window
================

* https://github.com/pothosware/pothos-gui/blob/master/MessageWindow/LoggerChannel.cpp

Currently, there is one logger window and its subscribed to all log messages. We can get very creative here and have multiple tabs for different log sources, or log levels. Or simply add an option to set the log level as a filter. There is an open issue to improve this FWIW: https://github.com/pothosware/pothos-gui/issues/8

Thoughts summary
================

Ultimately, I want to implement all 3 :-)

But I think it would be good if SoapySDR had a setLogLevel() call. And a default log level that can be modified with an environment variable. In addition, the SDR source and sink block should get an optional drop-down to set the SoapySDR::setLogLevel() that will modify it when specified.

A default log level for SoapySDR would also imply that the current debug message will be hidden by default until the API call to setLogLevel() is made or the env variable is set. I wonder if that will be confusing to users.

cjcliffe, 2015-10-21:
:+1: to setLogLevel() for me; bit flags or just a 'level' is fine too.   Typically ERROR and WARNING are the only types I see enabled by default for most applications.  I was going to just remove/hide those messages once it's fully polished but it's really nice to have the option for end-user debugging purposes.

Toontje, 2015-10-21:
+1 for the setLogLevel()
Agree with Charles that ERROR and WARNING are most commonly used, but we
need to be able to set DEBUG for, well, debugging purposes.

guruofquality, 2015-10-21:
Codes on master, give it a shot:

```
 * The default log level threshold is SOAPY_SDR_INFO.
 * Log messages with lower priorities are dropped.
 *
 * The default threshold can be set via the
 * SOAPY_SDR_LOG_LEVEL environment variable.
 * Set SOAPY_SDR_LOG_LEVEL to the string value:
 * "WARNING", "ERROR", "DEBUG", etc...
 * or set it to the equivalent integer value.
```

```
/*!
 * Set the log level threshold.
 * Log messages with lower priority are dropped.
 */
SOAPY_SDR_API void SoapySDR_setLogLevel(const SoapySDRLogLevel logLevel);
```

closed 2015-10-23 by guruofquality