pothosware/SoapySDR #41: Optional path parameter for loadModules(..) and status results.

cjcliffe, 2015-10-23:
It would be nice to get a result array from loadModules with module information including a pass/fail and any relevant message i.e. "failed ABI check" -- not everyone will have a console or be able to set an environment variable and I'd like a friendly way to mitigate this such as allowing them to pick the path or opting to point it at a bundled modules folder.  

loadModule() could benefit from returning a single result for adding and evaluating unknown module files supplied by the user so I could allow them to use/import them in an informed manner as well.

Cheers!

guruofquality, 2015-10-23:
Not a bad idea. The manual API control over modules and feedback is definitely something that's missing. I'm not 100% clear on the use case, so I don't want to go overboard with changes. I just want to make sure this other use-case is well supported. Here are a few thoughts:

loadModule()
===========

Because multiple registrations can exist inside a single module, the result of loadModule() should include two things 1) whether or not the module load itself failed and 2) a map of registration keys to error messages.

Besides the ABI checks, an additional error not yet checked for could be if two modules try to both register the same key. This could happen if there are duplicate modules from different versions, for example.

loadModules()
===========

So loadModules() is automatically called by Device::enumerate()/make() Once loadModules() is called once, it becomes a NOP on subsequent calls. This makes the behaviour nice and automatic for most apps, but it may interfere with someone trying to manually load individual modules.

Also, in terms of errors: Either there is a way to query an array of the results that loadModule() would have. Or in this use-case we don't use loadModules() in favor of manual control calling loadModule() one at a time.

unloading modules?
===========

Is there a need to unload modules? Currently, I assume that we load everything we can and just let the process exit unload everything. This will involve stashing the dlopen() handles along with whatever keys were registered when the module was loaded.

setting the search path?
===========

Do we really want to set the search path, or is this just a workaround for the automatic behaviour of loadModules(). Would being able to manually load all modules with loadModule(path) and disabling loadModules() automatic loading be a better, more controlled approach for managing modules in cubicsdr (for example). One thing I can do is make the listModules() function take a path, so the recursive directory search in Modules.cpp can be used in a client application.

cjcliffe, 2015-10-23:
The most common use case I'm guessing will be for people who did something like install SoapySDR modules through brew to /opt/local/ and I just want to specify another search path from the GUI instead of setting the SOAPY_SDR_ROOT env variable elsewhere.

How about default behaviour for loadModules() with no path and allow subsequent calls to loadModules(path) and only NOP if you call the same (or default) path again? 

Though if listModules(path) returns the module files and loadModule() returns a status (name conflict, incompatible ABI, etc.) it makes loadModules(path) mostly unnecessary and at least saves me from duplicating loadModules() functionality.

Unloading modules would be neat to have so that I could build and test updated modules on-the-fly or switch between multiple experimental modules for the same device but that's about the only use case I can think of for it; definitely not necessary though.  

If I could still load a module that had not loaded previously for some reason that would be helpful though -- so that users swapping in modules to find the working one can keep trying -- most likely a windows dll scenario trying to point it at multiple folders until it works..

guruofquality, 2015-10-25:
@cjcliffe Take a look at the "loader_work" branch.

* Added a listModules(path) that takes a path
* loadModule() returns an error msg
* added unloadModule() and returns an error msg as well
* getLoaderResult() returns individual registration errors for a given loaded module
* only the automatic loadModules() does any prints now (and through the logger)
* loadModules() also checks if a module is already loaded

cjcliffe, 2015-10-25:
Excellent, that sounds like it covers everything I need; will create a branch and try it out soon.  This will also make it easy to create a "bundled" module release for windows/mac that can still use external modules if available.

cjcliffe, 2015-10-30:
The listModules(path), loadModule(path) and subsequent loadModules() calls all appear to be working fine on windows -- just had to make one fix to a string cast to get it to compile with MSVC.  I've managed to make an experimental bundled version at https://github.com/cjcliffe/CubicSDR/releases/tag/v0.1.15-alpha that appears to be working ok here so far.  Haven't tested the return values yet but will be trying it soon once I implement the module refresh/reload/add path stuff in CubicSDR.

guruofquality, 2015-11-12:
Closing this issue. I think the changes were very sensible and helped to make the default use case better as well. Lets open a new one if something else comes up.

closed 2015-11-12 by guruofquality