pothosware/SoapySDR #43: command line stream utilities

guruofquality, 2015-10-25:
Expand SoapySDRUtil to provide stream testing and demonstration utilities:

* receive a timed burst
* transmit a timed burst
* receive to a file/or null
* transmit from file/or waveform

Options to provide:
* --frequency
* --clockRate (optional)
* --settings (default empty or dict)
* --tuneArgs (default empty or dict)
* --channel (default 0)
* --streamArgs (default empty or dict)
* --rate (sample rate)
* --gain (number or dict)
* --file (optional)
* --waveform (sine, constant)
* --nsamps (specify for finite burst)
* --time (time from now for timed burst)

ha7ilm, 2016-07-23:
Hi Josh,

Thanks for SoapySDR, it's great!

I started to work on this issue, because I wanted to use OpenWebRX with SoapySDR.
Some of the described functionality works:
https://github.com/ha7ilm/SoapySDR/commits
What I have now: --receive, --channel, --frequency, --samplerate, --buffersize

Curently the samples are written to stdout, and it works with OpenWebRX as well.

I would also need real-time control over the parameters (like center frequency), so I'm thinking about using stdin or a named pipe for control.

Phlat, 2016-11-27:
Hi. I have and SDRPlay on a Raspberry Pi 3. Your SoapySDR is installed (ha7ilm) with the sdrplay driver. How did I use it in the openwebrx config file?

ha7ilm, 2016-11-27:
@Phlat 
Right now this might be an easier way to go: https://github.com/krippendorf/SDRPlayPorts

ha7ilm, 2016-11-27:
By the way, the same weekend I was working on this feature, `rx_tools` was released, which was more complete than my implementation: 
http://www.rtl-sdr.com/rx_tools-rtl-sdr-command-line-tools-rtl_power-rtl_fm-rtl_sdr-now-compatible-with-almost-any-sdr/

Phlat, 2016-11-27:
Thank you. I have rx_tools installed. Is there an example using rx_sdr for openwebrx with the driver=sdrplay? I can't seem to figure it out.

ha7ilm, 2016-11-27:
@Phlat No example yet. Please then go with SDRPlayPorts as previously suggested.

Phlat, 2016-11-27:
@ha7ilm  It's now working with play_sdr.

73

closed 2022-11-24 by ncorgan