pothosware/SoapySDR #44: SoapyAudio module

cjcliffe, 2015-10-25:
I've got an idea to create a SoapyAudio module using RtAudio that will support input and output to sound cards using Real or Complex (stereo I/Q) for interfacing with VLF via mic connection or radio hardware that supports I/Q data.

Transmission of Real and I/Q data can be supported on line-out/speaker as well.

Additionally I/O from .WAV or other audio files will be supported through stream args / settings.

Could also be used as a general replacement for an audio driver; eliminating the need for built-in audio support.

If you can set up a repository for it I'll probably start hacking it together in the coming weeks :)

guruofquality, 2015-10-25:
I can see how making the audio abstraction behind the same API can be a big help. I think that some of these devices may also have a serial port for configuration/control. Perhaps those devices could have their own Soapy SDR module for the control, but internally they instantiate the SoapyAudio and just forward their stream calls. Just a thought.

I haven't use RtAudio. I have been using PortAudio for cross platform stuff. So if its of any use, here is the code for the audio source and sink block in pothos. It makes use of PortAudio for multiple data types and multiple channel streams vs interleaved data.

* https://github.com/pothosware/pothos-audio/wiki
* https://github.com/pothosware/pothos-audio/blob/master/AudioSink.cpp
* https://github.com/pothosware/pothos-audio/blob/master/AudioSource.cpp

cjcliffe, 2015-10-25:
RtAudio supports several different APIs well under the hood; whatever is the best for each platform/compile can be used automatically (including dummy driver).  It also includes support for JACK audio which allows advanced routing on Linux/OSX.

Unix/Linux in particular can use ALSA, OSS, PULSE and JACK (all simultaneously if desired); Windows has DirectSound or WASAPI and OSX has CoreAudio.

It's been the most stable for everthing I've thrown at it so far and I'm familiar with it enough to know I can make a full module out of it -- you can find more info here: https://github.com/thestk/rtaudio

guruofquality, 2015-10-25:
Have at it :-)
https://github.com/pothosware/SoapyAudio

closed 2015-10-25 by cjcliffe