pothosware/SoapySDR #45: Error building latest SoapySDR on OSX

Toontje, 2015-10-28:
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

API Version: v0.4.0-ge62dba9d
ABI Version: v0.4-2
Install root: /usr/local
Module found: /usr/local/lib/SoapySDR/modules/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules/librtlsdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
Loading modules... [ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libremoteSupport.so)
  null failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libremoteSupport.so)
  remote failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/librtlsdrSupport.so)
  rtlsdr failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0
done
Available factories...null, sdrplay,

cjcliffe, 2015-10-28:
Need to pull and rebuild SoapyRemote and SoapyRTLSDR now that 0.4.x is the minimum and ABI has changed.

guruofquality, 2015-10-28:
Better a print than a segfault. :-)

This error message is a little strange though. I may have to double check that logic:

> [ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libremoteSupport.so)
  null failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libremoteSupport.so)
  remote failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0

guruofquality, 2015-10-28:
> This error message is a little strange though. I may have to double check that logic:

Nothing wrong with the error message. It looks like the module load causes the older soapy sdr library to be loaded as well which also tries to register the "null" device in the older library. There is a better way to handle this. Null device is just used for unit tests so we have something to instantiate when hardware isnt there.

Toontje, 2015-10-28:
You wanted a segfault? Here you have a segfault! ;-)
Built SoapyRemote, then SoapyRTLSDR, then SoapySDR. On starting CubicSDR (no rebuild) i get a segfault on device enumeration:
```
SDR enumerator starting.
SoapySDR init..
	API Version: v0.4.0-ge62dba9d
	ABI Version: v0.4-2
	Install root: /usr/local
	Module found: /usr/local/lib/SoapySDR/modules/libremoteSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/librtlsdrSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
	Loading modules... done
	Available factories...null, remote, rtlsdr, sdrplay
[WARNING] SoapySSDPEndpoint failed join group udp://[ff02::c]:1900
  setsockopt(IPV6_ADD_MEMBERSHIP) [49: Can't assign requested address]
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0

mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...

mir_sdr_2500_Init: fnaddr = 6

mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
mir_sdr_SetFs: Sample Freq requested 2047999.978065
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2048000.0+0.0Hz=2048000.0Hz FsToggle->1
mir_sdr_SetRf: f->222064000.000Hz (int=25 frac=20 afc=0) fSynth:3553024000.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=222064000.0+0.0+0.0Hz+0.0Hz=222064000.0Hz RfToggle->1
mir_sdr_SetGr: GR->40[16,24,0,0] gRset->0x210 DCCALmode=4 DCCALspd=1 GrToggle->1
setToggleStates: initialising sampNum=0x0001fe5e, gainSetting=0x210, FsToggle=0, RfToggle=1, GrToggle=1
setToggleStates: initialising Fs=2047999.978, Rf=222064000.000, Gr=40
mir_sdr_2500_Readback001Trim: readback = 0xe30140d
mir_sdr_2500_Bypass001Trim: reg13=0x1ad reg14=0x6c014e
  driver = sdrplay
  label = SDRPlay RSP
Make device 0
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0

mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...

mir_sdr_2500_Init: fnaddr = 6

mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
mir_sdr_SetFs: Sample Freq requested 2047999.978065
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2048000.0+0.0Hz=2048000.0Hz FsToggle->1
mir_sdr_SetRf: f->222064000.000Hz (int=25 frac=20 afc=0) fSynth:3553024000.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=222064000.0+0.0+0.0Hz+0.0Hz=222064000.0Hz RfToggle->1
mir_sdr_SetGr: GR->40[16,24,0,0] gRset->0x210 DCCALmode=4 DCCALspd=1 GrToggle->1
setToggleStates: initialising sampNum=0x0001fe36, gainSetting=0x210, FsToggle=0, RfToggle=0, GrToggle=0
setToggleStates: initialising Fs=2047999.978, Rf=222064000.000, Gr=40
mir_sdr_2500_Readback001Trim: readback = 0xe30140d
mir_sdr_2500_Bypass001Trim: reg13=0x1ad reg14=0x6c014e
  mir_sdr_version=1.800000
[1]    3444 segmentation fault  ./CubicSDR
```

guruofquality, 2015-10-28:
@cjcliffe  not it -- I imagine this is from something inside of sdr play driver?

> [WARNING] SoapySSDPEndpoint failed join group udp://[ff02::c]:1900
  setsockopt(IPV6_ADD_MEMBERSHIP) [49: Can't assign requested address]

This is harmless BTW. But it does confirm the issue with the multicast ipv6 I was experiencing w/ OSX.

cjcliffe, 2015-10-28:
Just pulled and re-built everything here on OSX, seeing similar warnings but no segfaults from CubicSDR:

```
SDR enumerator starting.
SoapySDR init..
	API Version: v0.4.0-ge62dba9d
	ABI Version: v0.4-2
	Install root: /usr/local
	Module found: /usr/local/lib/SoapySDR/modules/libairspySupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libbladeRFSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libhackrfSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libremoteSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/librfspaceSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/librtlsdrSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
	Loading modules... [1m[31m[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libairspySupport.so)
  airspy failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0[0m
[1m[31m[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libairspySupport.so)
  null failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0[0m
[1m[31m[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libbladeRFSupport.so)
  bladerf failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0[0m
[1m[31m[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libhackrfSupport.so)
  hackrf failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0[0m
[1m[31m[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/librfspaceSupport.so)
  rfspace failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0[0m
done
	Available factories...null, remote, rtlsdr, sdrplay
[1m[33m[WARNING] SoapySSDPEndpoint failed join group udp://[ff02::c]:1900
  setsockopt(IPV6_ADD_MEMBERSHIP) [49: Can't assign requested address][0m
```

Only thing I see different in @Toontje 's log is here:
``` 
mir_sdr_usb_USB DLL: Revision 1.8.0
mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...
mir_sdr_2500_Init: fnaddr = 6
.. init completes ..
```

Where as mine reads:
```
mir_sdr_usb_USB DLL: Revision 1.8.0
mir_sdr_2500_Init: fnaddr = 5
.. init completes ..
```

cjcliffe, 2015-10-28:
@Toontje I just noticed you wrote:

"SoapyRemote, then SoapyRTLSDR, then SoapySDR"

You have to build and install SoapySDR first as the rest depend on the updated code and headers :)

Toontje, 2015-10-28:
Ok, ok. Done that. All working fine now. I keep having the below warning. I had IPV6 off, but now that i switched it on the warning doesn't go away.
```
[WARNING] SoapySSDPEndpoint failed join group udp://[ff02::c]:1900
  setsockopt(IPV6_ADD_MEMBERSHIP) [49: Can't assign requested address]
```

cjcliffe, 2015-10-28:
@Toontje yeah that's from the SoapyRemote module trying to register it's new auto-discovery to IPV6; going to be implementing that soon so that it can find SoapyRemote devices on the network :)

Toontje, 2015-10-28:
I'd prefer that you'd pick up the digital_lab branch again. ;-) But then again, it's not me who decides the priorities here. I am just a tester.

cjcliffe, 2015-10-28:
digital_lab will be integrated sooner than later there's just a ton of UI and DSP work to do to make it useful; at first I'm just aiming for a X/Y scope and hex/binary display -- it may be awhile before it's interactive.  Step 1 is integrating the digital_lab updates (even if hidden at first) and then cleaning up the demodulator code and adding some output for it; that's my next major clean-up priority -- guessing that will be before the end of the year and we'll be making in-depth issues about digital mode LUA scripting by this time next year ;)

Toontje, 2015-10-28:
You are doing great work, man!! Well appreciated.

cjcliffe, 2015-10-29:
@Toontje thanks; between your keen eye for detail, @guruofquality's superhuman efforts and the SDR and SDRPlay community everything is really coming together nicely :+1:

closed 2015-10-29 by cjcliffe