pothosware/SoapySDR #46: Java or Clojure binding support

BooBSD, 2015-10-29:
Hi guys,
Are you planning to realize Java (or maybe Clojure) binding support?
That would be very useful.
Thanks.

guruofquality, 2015-10-29:
I think more language bindings are a good idea.

Currently we are using swig for the python bindings. I don't love swig, but its doing a good job of automatically wrapping the many API calls automatically. Then there is some customization around the streaming API to deal with pointers, return by reference, and numpy arrays:  https://github.com/pothosware/SoapySDR/tree/master/python

I imagine that a very similar java/ directory could handle the java bindings (using swig and some custom code around the streaming to properly use java arrays).  And then clojure is covered by the java bindings? http://clojure.org/java_interop

guruofquality, 2015-11-04:
There's a pretty strait forward example of java + swig + cmake here: https://github.com/ptitpoulpe/cmake-swig-java-example It definitely has more boiler plate than I expected, and it needs some path fixes to handle out of tree builds... but otherwise it seems formulaic. If any one is interested in getting some basic swig (or something else) bindings working for java, I wouldn’t mind joining in to help polish it off, and work through some of the more complicated parts if needed.

ericek111, 2024-11-18:
Here's my attempt (WIP): https://github.com/ericek111/JSoapy