pothosware/SoapySDR #50: C#/.Net/Mono bindings

guruofquality, 2015-11-04:
We are interested in doing other language bindings for SoapySDR so that anyone can get involved in SDR using their favourite language, tools, and libraries. We may already have a volunteer to tackle C# bindings for SoapySDR, but first we need to know if there is any interest. Please let us know in the comments. Thanks!

ooeygui, 2016-02-09:
:+1: Would also be interested in a UWP Component for Windows 10.

cjam, 2017-01-24:
Would be interested in using C# mono bindings in Xamarin.

Ronkpa, 2017-04-25:
Yes I'm interested in C# interface to SoapySDR (and in turn to SDRPlay RSP2).

OhSoGood, 2018-03-30:
@guruofquality , do you know if a c# has been created for soapy? Or a java one (which should be close)? I've searched but found nothing for soapy itself.

I saw there was a semi-old project of java binding for Pothos also, but I'd be more interested in the soapy layer.

guruofquality, 2018-03-30:
Not that I know of. Given that SoapySDR bindings for python are basically swig based, I think we could get away with another swig .i file and some build glue for C# or java (which I do think are similar). It would be a lot simpler than the way Pothos handles bindings (because he calls into the language from c++ runtime). Any volunteers? :-)

M0LTE, 2021-12-28:
Yes, I’m interested in calling the LimeSDR Mini from C#. Target platform will be .NET 6 on Raspberry Pi 4. Did the SWIG file above ever get tested? Did any bindings ever get generated? Did the API change much since the file above was written?

OhSoGood, 2021-12-28:
Hi @M0LTE , our needs changed in the meantime and we went coding directly in C, so I didn't go further on the C# binding. What I had done looked promising, but I never tested it. As the API seldom changes (almost never I'd say), I encourage you to build up on it and to share with others.

ncorgan, 2021-12-28:
https://github.com/pothosware/SoapySDR/tree/wip/csharp

I have .NET bindings in the works, but due to CMake (and sanity) limitations, it's only buildable with MSVC.

It's mostly done, and I don't anticipate too much C#-specific API churn, but no guarantees on that.

Hagtronics, 2021-12-29:
ncorgan - Thanks for the timely update - I am working on a few projects now with Pluto and SDRPlay devices and it would be good for me to have the same 'hook' to them via Soapy. Right now I am using their respective DLL's. I (we) appreciate your work on this. I use C# because its: Fast, easy, powerful, and reliable. Plus it's supported and cost effective.

ncorgan, 2022-11-24:
Closing, support was recently merged in master.

closed 2022-11-24 by ncorgan