pothosware/SoapySDR #54: Arch linux packaging

guruofquality, 2015-11-14:
Looking for volunteers interested in creating/testing packages for arch linux. Looks like we need a handful of PKGBUILD files very similar to (https://github.com/pothosware/homebrew-pothos/). We can have a stand-alone arch package repo as well; whatever is best.

Thoughts, volunteers? Let us know in the comments.

tomswartz07, 2015-11-18:
I have a rough cut (read: minimally tested) PKGBUILD for Arch here: https://github.com/tomswartz07/CubicSDR-PKGBUILDS/blob/master/soapysdr/PKGBUILD

I'd be more than happy to help with the 'official' PKGBUILD file.

guruofquality, 2015-11-19:
Is there any official naming convention here with the repos: Should I just make PKGBUILDS.git?

tomswartz07, 2015-11-19:
Well, there are two options:
If you want to keep your PKGBUILDs separate on GitHub, you can name them whatever you want.

When the package is actually submitted to the AUR, you'll need to add their repository as a remote.
It's all documented here: https://wiki.archlinux.org/index.php/Arch_User_Repository#Submitting_packages

guruofquality, 2015-11-21:
Well, I'm making myself an arch vm, and you now should have free reign over the new repo: https://github.com/pothosware/PothosSDR-PKGBUILDS

FFY00, 2019-06-01:
I think this can be closed as I moved pretty much every SoapySDR module to the official repos.

guruofquality, 2019-06-01:
cool!

closed 2019-06-01 by guruofquality