pothosware/SoapySDR #63: /usr/local/include/SoapySDR/Device.h:60:1: error: unknown type name ‘SoapySDRDevice’ (Using C api)

jazzkutya, 2016-03-14:
Hi, i want to learn using SoapySDR with C, but I'm completely blocked by this error.
The source I'm trying to compile with SoapySDR is here: https://github.com/jazzkutya/learnsoapysdr

jazzkutya, 2016-03-14:
Okay I've noticed that if I compile my stuff with g++ it compiles well. Also I read on http://stackoverflow.com/questions/8096887/mixing-c-and-c-with-cmake that the c source file having main() should be compiled with a c++ compiler. Now let's assume one is having the SoapySDR related code in a separate object and C file and uses a C compiler for that as it has no main(). In this case the problem can be fixed with changing in include/SoapySDR/Device.h:

struct SoapySDRDevice;
struct SoapySDRStream;

change these to lines to:
typedef struct SoapySDRDevice SoapySDRDevice;
typedef struct SoapySDRStream SoapySDRStream;

Should this change be committed to SoapySDR?

guruofquality, 2016-03-14:
This commit on the cfixes branch should do the trick: https://github.com/pothosware/SoapySDR/commit/7f8d0c93df28bf8039bcb85b5f08630d0335a2c6

thanks for reporting

guruofquality, 2016-03-14:
Also, your suggested fix is actually nicer, I made the change. :-)

jazzkutya, 2016-03-14:
thanks, it's working :)

jazzkutya, 2016-03-14:
just a small note, i wrote the wrong url here for info on mixing C and C++ code, the correct is https://isocpp.org/wiki/faq/mixing-c-and-cpp

guruofquality, 2016-03-14:
thanks, cfixes branch merged, closing

closed 2016-03-14 by guruofquality