pothosware/SoapySDR #64: segfault in SoapySDRDevice_enumerate()

jazzkutya, 2016-03-14:
Hi,

I get a segfault in both SoapySDRDevice_enumerate and SoapySDRDevice_enumerateStrArgs while trying to use the C api.
Using the C++ api is OK.

C code that gives segfault: soppy.c at https://github.com/jazzkutya/learnsoapysdr
C++ code that is working OK: soppy++.cpp in the same github repo.

```
vrm@zivatar:~/mygit/learnsoapysdr/build$ ltrace ./soppy
__libc_start_main(0x4006b0, 1, 0x7ffe662d14f8, 0x4007d0 <unfinished ...>
SoapySDRDevice_enumerate(0x7ffe662d13f0, 0x7ffe662d13e0, 0x7ffe662d1508, 5mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

Warning: libusb_claim_interface() -6

mir_sdr_2500_Init: fnaddr = 8

mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000005
initHw: Tuner Register0 = 0x04f440
mir_sdr_SetFs: Sample Freq requested 2047999.978065
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2048000.0+0.0Hz=2048000.0Hz FsToggle->1
mir_sdr_SetRf: f->222064000.000Hz (int=25 frac=20 afc=0) fSynth:3553024000.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=222064000.0+0.0+0.0Hz+0.0Hz=222064000.0Hz RfToggle->1
mir_sdr_SetGr: GR->40[16,24,0,0] gRset->0x210 DCCALmode=4 DCCALspd=1 GrToggle->1
setToggleStates: initialising sampNum=0x0001f802, gainSetting=0x210, FsToggle=0, RfToggle=1, GrToggle=1
setToggleStates: initialising Fs=2047999.978, Rf=222064000.000, Gr=40
mir_sdr_2500_Readback001Trim: readback = 0xe30160f
mir_sdr_2500_Bypass001Trim: reg13=0x1ed reg14=0x6c016e
 <no return ...>
--- SIGSEGV (Segmentation fault) ---
+++ killed by SIGSEGV +++
```

guruofquality, 2016-03-14:
Should be fixed in the cfixes branch, here is the commit: https://github.com/pothosware/SoapySDR/commit/6b1b9182a73c0d48e1fd932221ff2c61270eddad

jazzkutya, 2016-03-14:
Thanks, it's working

guruofquality, 2016-03-14:
merged, closing

closed 2016-03-14 by guruofquality