pothosware/SoapySDR #65: Additional ways to specify module search path

guruofquality, 2016-03-22:
The goal: To make it easier for 3rd parties to install hardware support modules and to have them loaded automatically at runtime:

* On windows, a registry path that is searched for keys. This is very similar to how the python path registry entry works. Installers for 3rd party support modules can create their own unique key in the registry and remove it on uninstall.

* I assume most unix style installers will use the lib/SoapySDR/modules directory to install support modules. Consider an etc conf.d style directory much like the registry option above where installers can deposit their own search paths into a file like /etc/SoapySDR/conf.d/MySuperSDR.conf where MySuperSDR might be a package that goes into /opt and not the regular file system hierarchy.

* A conf/ini style file in the HOME/APPDATA directory that the user can edit to specify additional search paths.

ast, 2018-05-21:
What I would like is an env variable where you can add a search path/paths. This is very useful for plugin development, ie. Gstreamer has GST_PLUGIN_PATH that you can set to your build directory.

guruofquality, 2018-05-21:
yes: https://github.com/pothosware/SoapySDR/blob/master/lib/Modules.in.cpp#L141 SOAPY_SDR_PLUGIN_PATH

albestro, 2021-04-11:
I had the same concern and I found out by looking at the code, but then before updating the wiki I found out this issue.

I don't know what's the policy for updating the wiki.

Specifically I would add in the section https://github.com/pothosware/SoapySDR/wiki#plugin-architecture something like

```
Via the environment variable `SOAPY_SDR_PLUGIN_PATH` it is possible to specify custom alternative
search paths for installed modules.
(e.g. `export SOAPY_SDR_PLUGIN_PATH=~/workspace/SoapyRemote/build/install_here/lib/SoapySDR/modules0.8`)
```

guruofquality, 2021-04-12:
> I don't know what's the policy for updating the wiki

Go for it :-) Sounds good to me

closed 2022-11-24 by ncorgan