pothosware/SoapySDR #67: MSVC14 issue?

cjcliffe, 2016-03-30:
Noticing that latest SoapySDR is possibly setting individual CMAKE_INSTALL_PREFIX per module.

Not sure if I did something weird but when I fully cleaned and re-generated all the cmake configurations and ran my usual build/install script I ended up with folders like:

```
C:\Program Files\SoapyAudio
C:\Program Files\SoapySDRPlay
C:\Program Files\SoapyHackRF
```

etc.. which matched the CMAKE_INSTALL_PREFIX generated by default in each.

Setting CMAKE_INSTALL_PREFIX for each build to C:\Program Files\SoapySDR resolves the issue but I don't remember having to do this before.

guruofquality, 2016-03-30:
The SOAPY_SDR_MODULE_UTIL() installed files into the same root as soapysdr, but this had some problematic implications which clashed with at least brew, and any situation where you can imagine a permission issue and/or different install paths: https://github.com/pothosware/SoapySDR/issues/60

So, I was thinking that I probably shouldn't do that. Sorry for the breakage though. Setting CMAKE_INSTALL_PREFIX to match for all projects is probably the best way to handle this. That's also what I do for the PothosSDR build.

Anyway, having a ton of different install paths should be a legit thing. Imagine 3rd party installers that are not only built against a PothosSDR install, but they install into their own root path in program files. Hence this issue as well: https://github.com/pothosware/SoapySDR/issues/65

guruofquality, 2016-04-05:
I think we are good. Sorry for the trouble!

closed 2016-04-05 by guruofquality