pothosware/SoapySDR #68: UDP Error on OS X

mehdideveloper, 2016-05-12:
Hi
When I run "SoapySDRUtil --find" with my SDRPlay connected to my MacBook Pro, it finds the SDRPlay and prints it information, but it gives a UDP error too.
How can I fix it? 
Output of command:

```
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [49: Can't assign requested address]
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0
...
```

Thanks
Mehdi

guruofquality, 2016-05-12:
Just for info, [SoapyRemote](https://github.com/pothosware/SoapyRemote/wiki) uses [SSDP](https://en.wikipedia.org/wiki/Simple_Service_Discovery_Protocol) to discover devices on the network available through SoapyRemote. The underlying SSDP code tries to find the device over both IPv4 and IPv6.

In this case, it looks like this this case it failed to send a request packet to the IPv6 multicast address for SSDP. I don't see an error for IPv4, -- so basically its working.  I have no idea why IPv6 failed in the sendto call, but not the earlier bind, group join, etc calls that setup the socket. Perhaps a firewall or network config issue.

So I wouldn’t worry. But out of curiosity: What was the client machine? and specifically what OS version?

mehdideveloper, 2016-05-12:
I'm using OS X 10.11.4 on MacBook Pro 13" 2015
I disabled the IPv6 with "networksetup -setv6off Wi-Fi"
Now I don't get that error anymore!
For the record, the internal OS X Firewall is off, but I've installed Little Snitch.
I don't have a problem and it works, but I thought you should know about this issue and I'm more than happy to help troubleshoot it.

guruofquality, 2016-05-12:
Thats @mehdideveloper thats a good data point. The reason I asked about the OS and stuff is that apple has occasionally turned out to be a little different when it comes to setting up multicast groups, and its possible that something changed since I last tested things (Yosemite).

I cant debug this until I update my test vm. Not to be overwhelming, but if you are curious to mess around here are some pointers:

* The entire SSDP implementation is located in this file: https://github.com/pothosware/SoapyRemote/blob/master/common/SoapySSDPEndpoint.cpp
* And all of the socket calls are encapsulated here: https://github.com/pothosware/SoapyRemote/blob/master/common/SoapyRPCSocket.cpp

Notice some ``#ifdef __APPLE__`` code here that gets called by ``SoapyRPCSocket::multicastJoin``:
https://github.com/pothosware/SoapyRemote/blob/master/common/SoapyRPCSocket.cpp#L184

My thought is that somehow the socket setup with multicastJoin() isnt quite right and the call to sendto() in the SSDP code fails. Perhaps  getDefaultIfaceIndex() is returning the wrong thing or multicastJoin() needs to be tweaked in some way.

Its also entirely possible that the address passed to SoapyRPCSocket::sendto() is parsed into complete junk for some reason. That doesnt seem quite right since this exact call has worked in other places and other times, but thats one possible interpretation of the "49: Can't assign requested address".

mehdideveloper, 2016-05-13:
It's very strange @guruofquality !
After multiple times of turning IPv6 on and off, for testing your code, I no longer get the error!
I tested "getDefaultIfaceIndex" function. This function always finds the "loIface" interface correctly. But the "enIface" is dependent on whether IPv6 is disabled or not.
When IPv6 is enabled, this function returns the index of my "en0" (Wi-Fi interface), but when I disable IPv6, it returns "awdl0" (Apple Wireless Direct Link interface)
Anyway, I don't get this error anymore and it's weird. (I haven't changed anything other than enabling and disabling IPv6 for testing)

guruofquality, 2016-05-13:
@mehdideveloper Thanks for checking. I guess it was just a temporary network oddity. Let me know if it comes back up.

closed 2016-05-13 by guruofquality