pothosware/SoapySDR #69: Question about "SoapySDRUtil" strange behavior

mehdideveloper, 2016-05-13:
Hi
I don't know whether it's a bug or not.
I've compiled SoapySDR (latest) from git and it's working, but there is a strange thing happening.
When no SDR is attached to my laptop, if I run "SoapySDRUtil --find", it returns this:

```
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1
No devices found!
```

And this is the output of "SoapySDRUtil --info":

```
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

API Version: v0.5.0-gaffeccd2
ABI Version: v0.5-1
Install root: /usr/local
Module found: /usr/local/lib/SoapySDR/modules/libremoteSupport.so
Module found: /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
Loading modules... done
Available factories...null, remote, sdrplay,
```

From what I understand, every time we want to find which devices are present on our system, we have to call and load their driver. Can't we just enumerate devices based on other information? (enumerate USB devices and if we found a match, we can load the driver and print the information)
So if I had more factories available, SoapySDRUtil would run and load all of them just to see whether they're connected to the system or not? It does not seem very logical to me.
What am I missing here?

Thanks
Mehdi

guruofquality, 2016-05-13:
I think the missing detail is the optional arguments which can be used to filter what drivers are searched as well as within devices supported by a given driver.  ``SoapySDRUtill --find`` will call the registered discovery function for all supported drivers and return all of the results. However, ``SoapySDRUtill --find="driver=sdrplay"`` will find only those devices supported by the sdrplay driver.

Those same arguments are passed to the driver's discovery function so the driver can perform additional filtering at the device level. For example, many devices that have serials will support filtering by serial number. ``SoapySDRUtill --find="serial=abcdxyz"`` would only find one board. Or do something like ``SoapySDRUtill --find="driver=foo,serial=abcdxyz"`` to make sure that only one particular driver was ever searched for this serial.

Since you are using SoapyRemote, you might find more details about this here as well: https://github.com/pothosware/SoapyRemote/wiki#remote-device-args

mehdideveloper, 2016-05-13:
OK, so it's an SDRPlay factory/driver problem and not SoapySDR's.
Thanks

guruofquality, 2016-05-16:
If you are talking about the "mir_sdr_Init" prints, it looks like the underlying "mirsdrapi-rsp" library is very talkative since this isnt coming from SoapySDRPlay itself. Basically whenever the SoapySDRPlay is invoked to find devices, its going to run the init code and cause these prints. There isnt anyway around it. In any case, thanks for the report!

closed 2016-05-16 by guruofquality