pothosware/SoapySDR #71: Add SoapySDR to MacPorts

guruofquality, 2016-05-26:
I've been maintaining the relevant homebrew recipes. But I'm not very familiar with MacPorts. I don't believe that there is a SoapySDR port for MacPorts. Also, I just leaned that the gr-osmosdr port doesn’t currently build with SoapySDR support, forcing users to rebuild that package manually.

I would be grateful if anyone familiar with MacPorts could help out with this. I don't think we necessary need ports for all of the SoapySDR support modules, just SoapySDR itself so dependent apps like gr-osmosdr can use it.

gedger, 2016-07-09:
I now have SoapySDR working on MacPorts by modifying the Portfile.

`sudo port install gr-osmosdr`

Edit the Portfile

```
cd $(port dir gr-osmosdr)
sudo vi Portfile
```

Add the following at the end of the file

```
variant soapy description "Install ${name} with support for SoapySDR" {
    configure.args-append -DENABLE_SOAPY=ON
}

if {![variant_isset soapy]} {
    configure.args-append -DENABLE_SOAPY=OFF
}
```

then rebuild the Port enabling soapy

`sudo port install gr-osmosdr +soapy`

This should now enable SoapySDR support in gr-osmosdr

If this works for others then I'll see what you have to do to submit to Macports as a permanent change.

Richard

gedger, 2016-07-09:
I now have a draft Portfile for SoapySDR. It seemed very easy which makes me think I've missed something....Just need to confirm how versions are going to be tagged in git, Once #77 is complete then should be able to complete and submit to MacPorts. 

```
# -*- coding: utf-8; mode: tcl; tab-width: 4; indent-tabs-mode: nil; c-basic-offset: 4 -*- vim:fenc=utf-8:filetype=tcl:et:sw=4:ts=4:sts=4
# $Id$

PortSystem          1.0
PortGroup           make 1.0

name                soapysdr
maintainers       gedger

description         A fresh and clean vendor neutral and platform independent SDR support library
long_description    SoapySDR is an open-source generalized C/C++ API and runtime library for interfacing with SDR devices. With SoapySDR, you can instantiate, configure, and stream with an SDR device in a variety of environments. Both osmosdr and uhd devices are available within SoapySDR. In addition, vendors can directly support their hardware using SoapySDR device modules. There are wrappers for both gr-osmosdr, uhd, and gr-uhd to bring an ecosystem of existing applications to SoapySDR devices. And SoapySDR has support for powerful platforms like GNU Radio and Pothos.

categories          science
homepage            https://github.com/pothosware/SoapySDR/wiki
license             Boost Software License V1.0
platforms           darwin machos

version             current

master_sites        https://github.com/pothosware/SoapySDR

fetch.type          git
git.url		     https://github.com/pothosware/SoapySDR.git

```

guruofquality, 2016-07-10:
@gedger Thanks a lot. I should have tags out for SoapySDR and SoapyRemote this weekend.

gedger, 2016-07-10:
Having looked at this a bit more I think you need to do a git release, this may tag it at the same time? I've also now have Portfiles for soapyremote and working on one for soapysdrplay, which is a bit more problematic for some reason....

guruofquality, 2016-07-10:
Since there were API additions to SoapySDR, SoapyRemote also had support for these additions. So they get separate tags for patch releases, but in terms of a API-level release like this one, both are getting tagged roughly at the same time.

-> Just noting that SoapyRemote is backwards compatible (much ifdefs), so you shouldn’t have any compilation issues with an older SoapySDR or vice-versa.

gedger, 2016-07-17:
Away at the moment,  I will submit to MacPorts when I return.

gedger, 2016-08-05:
I have submitted Portfile for SoapySDR, once this gets accepted I'll add SoapyRemote 

https://trac.macports.org/ticket/51940

gedger, 2016-08-25:
Now accepted, port SoapySDR should be live....

guruofquality, 2016-08-26:
Sweet! Much appreciated. Feel free to leave this open to track other mac ports endeavours, or open new issues if need be.

closed 2017-01-28 by guruofquality

jerrynsr, 2017-09-10:
I had a problem with port install gmp.    I figured it out.  I have a very fast machine so I enabled ccache.  Well guess what.. when you have 32 parallel compiles going ccache screws up big time.
You have to disable it if you have a very fast machine it will cause your builds to fail.. even stupid configure errors.