pothosware/SoapySDR #82: testsuite tests installed, not built library

andreasbombe, 2016-07-02:
The test programs run with `make test` link against the installed library instead of the just built library. This also means that if there is no library already installed, the tests will fail.

Since `debhelper` automatically runs `make test`, building Debian packages directly from a git clone is impossible without building and installing the library first (or disabling the tests altogether).

guruofquality, 2016-07-02:
So I dont currently run the unit tests in the debian/rules, and the travis build stuff runs them after the install. However, I think this should work, but it depends on the CMAKE RPATH setup. I know that in my checked in debian/rules, the -DCMAKE_SKIP_RPATH=TRUE is setup and so without any RPATH the tests are either going to fail or use the installed library I guess. But I'm curious what if any CMAKE_\*RPATH\* variables were being set in this case? Because I think the default has build RPATH so the tests will work before the library is installed.

guruofquality, 2016-07-03:
Update: I overrode the tests for now in rules until the RPATH is figured out correctly. I think the RPATH setup changed since the last release for some mac ports request and I didnt realize that the tests were run or that they were now failing. I also updated the soversion for the library. I will probably be tagging very soon. So its good to get this covered first. Thanks!

guruofquality, 2016-07-03:
I figured it out. The [snippet here from always full rpath](https://cmake.org/Wiki/CMake_RPATH_handling#Always_full_RPATH) doesn’t deal with the multi-arch well. So I just added another option to override this, and also re-enabled the build rpath and the unit test: https://github.com/pothosware/SoapySDR/commit/cbb0f7ede618bfce407108effb1237ee0b1aa0f7

closed 2016-07-03 by guruofquality