pothosware/SoapySDR #84: Frequency corrections API?

guruofquality, 2016-07-09:
I was thinking about a setFrequencyCorrections() call that fits in with the Frontend corrections family of API calls. RTLSDR for example has a frequency corrections API call in units of PPM that can be used to correct the tuning. We have exposed this via the setFrequency(..., "CORR", value) call, which isnt quite right because that call expects Hz. I'm wondering how many other devices offer some kind of fine tuning corrections, what the units are typically for these calls, and if this merits an API addition -- which might be more straightforward and beneficial to more than one device.

romeojulietthotel, 2017-04-03:
Yes, agreed, there would be a benefit to have such an API. But I don't know what units the other devices expect.

Finding what units each device expects requires some investigation. Maybe someone reading this could do it easily. It would take some time for me to contribute. I will try finding that info for one device and see how hard that is for a start.

guruofquality, 2017-04-20:
Implemented and added feature to various projects where applicable. New tags for 0.6 are in the works.

closed 2017-04-20 by guruofquality