pothosware/SoapySDR #87: Functions to convert between Kwargs and CSV style markup

guruofquality, 2016-07-22:
This has come up more than once, but basically there is a need to convert between the ubiquitous "CSV key=value" string markup and the Kwargs string mapping. SoapySDR does have a markup to Kwargs internally to handle the device constructor from string. But it may be useful for others to provide this as an API utility function for C and C++:

Something like:
```
Kwargs args = stringToKwargs("driver=rtlsdr, serial=12345");

std::string markup = kwargsToString(args);
```

rxseger, 2016-07-22:
👍 would definitely find this useful, an initial implementation in https://github.com/rxseger/rx_tools/blob/master/src/convenience/convenience.c#L548 if you want to borrow it for SoapySDR but it has some limitations (no quoting/escaping so it cannot parse keys/values containing commas or equal signs, not sure how much that matters though).

guruofquality, 2016-07-22:
The escaping has not come up yet, but I think thats probably a good idea. The factory already has [this function internally](https://github.com/pothosware/SoapySDR/blob/master/lib/Factory.cpp#L72) without the non-standard C string calls. So I will probably go with that and wrap it into the C Types.h header as well.

closed 2016-08-23 by guruofquality