pothosware/SoapySDR #91: ambiguous naming scheme and documentation

lucasduffey, 2016-08-14:
I was trying to figure out what datatypes were returned by SoapySDR::Device::enumerate and noticed a couple issues.

1. "SoapySDR::Device::enumerate" is in "lib/Factory.cpp" instead of "lib/Device.cpp".
2. docs @ http://pothosware.github.io/SoapySDR/doxygen/0.5.1/classSoapySDR_1_1Device.html for "KwargsList SoapySDR::Device::enumerate" do not indicate what datatypes the individual devices are.

Reading the source for "SoapySDR::Device::enumerate", I'm still a little unsure.

lucasduffey, 2016-08-14:
when I figure this out more, I might try to contribute something useful

guruofquality, 2016-08-14:
> "SoapySDR::Device::enumerate" is in "lib/Factory.cpp" instead of "lib/Device.cpp".

I guess its a little inconsistent with the naming, but its not really a problem. Its just that the factory calls for discovery and instantiation all share data structures so they go in the same file.

> docs @ http://pothosware.github.io/SoapySDR/doxygen/0.5.1/classSoapySDR_1_1Device.html for "KwargsList SoapySDR::Device::enumerate" do not indicate what datatypes the individual devices are.

It returns a list of key/value maps, where each element of the list can be passed to Device::make() to instantiate a device handle.

closed 2016-09-14 by guruofquality