pothosware/SoapySDR #98: Version 0.5.2 does not compile on Raspbian Jessie

ISSchuster, 2016-08-23:
I can't compile version 0.5.2 on Raspbian Jessie - 0.5.1 is working fine...

root@raspberrypi:~/SoapySDR/build# make -j4
Scanning dependencies of target docs
Scanning dependencies of target SoapySDR
[  3%] [  7%] [ 11%] [ 14%] Generating documentation with doxygen
Building CXX object lib/CMakeFiles/SoapySDR.dir/Device.cpp.o
Building CXX object lib/CMakeFiles/SoapySDR.dir/Factory.cpp.o
Building CXX object lib/CMakeFiles/SoapySDR.dir/Registry.cpp.o
Warning: ignoring unsupported tag `GROUP_NESTED_COMPOUNDS =' at line 351, file /root/SoapySDR/build/docs/Doxyfile
Warning: ignoring unsupported tag `HIDE_COMPOUND_REFERENCE=' at line 522, file /root/SoapySDR/build/docs/Doxyfile
Warning: ignoring unsupported tag `WARN_AS_ERROR          =' at line 746, file /root/SoapySDR/build/docs/Doxyfile
Warning: ignoring unsupported tag `LATEX_EXTRA_STYLESHEET =' at line 1700, file /root/SoapySDR/build/docs/Doxyfile
Warning: ignoring unsupported tag `LATEX_TIMESTAMP        =' at line 1767, file /root/SoapySDR/build/docs/Doxyfile
Warning: ignoring unsupported tag `RTF_SOURCE_CODE        =' at line 1833, file /root/SoapySDR/build/docs/Doxyfile
Warning: ignoring unsupported tag `PLANTUML_INCLUDE_PATH  =' at line 2367, file /root/SoapySDR/build/docs/Doxyfile
Searching for include files...
Searching for example files...
Searching for images...
Searching for dot files...
Searching for msc files...
Searching for dia files...
Searching for files to exclude
Searching for files to process...
Searching for files in directory /root/SoapySDR/include/SoapySDR
Reading and parsing tag files
Parsing files
Preprocessing /root/SoapySDR/include/SoapySDR/Config.h...
Parsing file /root/SoapySDR/include/SoapySDR/Config.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Config.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Config.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Constants.h...
Parsing file /root/SoapySDR/include/SoapySDR/Constants.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Device.h...
Parsing file /root/SoapySDR/include/SoapySDR/Device.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Device.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Device.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Errors.h...
Parsing file /root/SoapySDR/include/SoapySDR/Errors.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Errors.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Errors.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Formats.h...
Parsing file /root/SoapySDR/include/SoapySDR/Formats.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Formats.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Formats.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Logger.h...
Parsing file /root/SoapySDR/include/SoapySDR/Logger.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Logger.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Logger.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Modules.h...
Parsing file /root/SoapySDR/include/SoapySDR/Modules.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Modules.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Modules.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Registry.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Registry.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Time.h...
Parsing file /root/SoapySDR/include/SoapySDR/Time.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Time.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Time.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Types.h...
Parsing file /root/SoapySDR/include/SoapySDR/Types.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Types.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Types.hpp...
Preprocessing /root/SoapySDR/include/SoapySDR/Version.h...
Parsing file /root/SoapySDR/include/SoapySDR/Version.h...
Preprocessing /root/SoapySDR/include/SoapySDR/Version.hpp...
Parsing file /root/SoapySDR/include/SoapySDR/Version.hpp...
Building group list...
Building directory list...
Building namespace list...
[... 201 lines cut ...]
                      ^
/root/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:5905:23: error: redefinition of âstruct swig::traits_asval<unsigned int>â
   template <>  struct traits_asval<unsigned int > {
                       ^
/root/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:5646:23: error: previous definition of âstruct swig::traits_asval<unsigned int>â
   template <>  struct traits_asval<size_t > {
                       ^
/root/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:5911:23: error: redefinition of âstruct swig::traits_from<unsigned int>â
   template <>  struct traits_from<unsigned int > {
                       ^
/root/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:5652:23: error: previous definition of âstruct swig::traits_from<unsigned int>â
   template <>  struct traits_from<size_t > {
                       ^
/root/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:5921:22: error: redefinition of âstruct swig::traits<std::vector<unsigned int> >â
  template <>  struct traits<std::vector<unsigned int, std::allocator< unsigned int > > > {
                      ^
/root/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:5662:22: error: previous definition of âstruct swig::traits<std::vector<unsigned int> >â
  template <>  struct traits<std::vector<size_t, std::allocator< size_t > > > {
                      ^
python/CMakeFiles/_SoapySDR.dir/build.make:62: recipe for target 'python/CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o' failed
make[2]: *** [python/CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:380: recipe for target 'python/CMakeFiles/_SoapySDR.dir/all' failed
make[1]: *** [python/CMakeFiles/_SoapySDR.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

guruofquality, 2016-08-23:
@ISSchuster Can you confirm what the branch is because there are no swig or API changes to cause this between 0.5.1 and 0.5.2... However, if you were building **master**, we just added a swig ``%template`` for some new API calls that use``std::vector<unsigned>``, and I think that on some 32-bit platforms size_t is a typedef of unsigned int -- which might cause this issue. Which means that there is a fix to master, but I dont think its for the release tag.

ISSchuster, 2016-08-23:
OK - my fault... It's the Master... I used "git clone https://github.com/pothosware/SoapySDR.git"

guruofquality, 2016-08-24:
FYI - This should fix it in the master branch: https://github.com/pothosware/SoapySDR/commit/91aa441fe8aba20027da2da0b2ac2018391190f6

closed 2016-08-24 by guruofquality