pothosware/SoapySDR #104: Multiple SDRPlays Weirdness

ehidle, 2017-01-07:
I'm having problems with multiple SDRPlay devices, especially when using in conjunction with RTL devices. (Compiled Soapy and modules for SDRPlay and RTLSDR this morning from git)

I have an RSP1, and RSP2, and two RTLs plugged in, and this is what I get. It sees both of the RSPs but only concludes that the RSP2 is an actual device. The RSP1 is invisible to soapy-aware apps (CubicSDR in this case). If I plug in a second RSP1 (total two RSP1, one RSP2, two RTLs), then I get one of the RSP1s to show, but the other is invisible. I can remove either of the RSP1s and go back to the first condition.

######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
fwDownload: FW image size = 6008
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0003 bus=001 port=003 devAvail=0
mir_sdr_usb_GetDevices Dev1:vid=1df7 pid=3010 rev=0207 serno=160506EF00 bus=002 port=002 devAvail=1
Found device 0
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T

Found device 1
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 99999999
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 1
  serial = 99999999
  tuner = Rafael Micro R820T

Found device 2
  driver = sdrplay
  label = SDRplay Dev1 RSP2 160506EF00

If I add a second RSP1 device, it will see one RSP2 and one RSP1, but the other RSP1 is invisible:

######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
fwDownload: FW image size = 6008
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0004 bus=001 port=004 devAvail=1
mir_sdr_usb_GetDevices Dev1:vid=1df7 pid=3010 rev=0207 serno=160506EF00 bus=002 port=002 devAvail=1
mir_sdr_usb_GetDevices Dev2:vid=1df7 pid=2500 rev=0206 serno=B0001P0003 bus=001 port=003 devAvail=0
Found device 0
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T

Found device 1
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 99999999
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 1
  serial = 99999999
  tuner = Rafael Micro R820T

Found device 2
  driver = sdrplay
  label = SDRplay Dev0 RSP1 B0001P0004

Found device 3
  driver = sdrplay
  label = SDRplay Dev1 RSP2 160506EF00

guruofquality, 2017-01-08:
It could be a bug in the sdr play libs, as far as SoapySDRPlay is concerned, its just reporting the result of mir_sdr_GetDevices(). For whatever reason mir_sdr_GetDevices() debugging prints out that it finds all of the devices but does not report it correctly in the resulting structures. You might need to let sdr play know about it

* https://github.com/pothosware/SoapySDRPlay/blob/master/Registation.cpp#L50

SDRplay, 2017-01-09:
@ehidle What platform are you seeing this on?

ehidle, 2017-01-09:
Ubuntu 16.04.1 LTS \n \l

Linux beethoven 4.4.0-57-lowlatency #78-Ubuntu SMP PREEMPT Sat Dec 10 01:37:35 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

-rw-r--r-- 1 root root 437960 Jan  7 13:02 /usr/local/lib/libmirsdrapi-rsp.so.2.09

SoapySDR and SDRPlay module pulled from git and built Jan 7 as well. 

Let me know if I can provide anything else that'd be helpful. Big thanks!!

ehidle, 2017-01-09:
PS I've read about the module conflict with some Linux kernels, but just wanted to add that there do not appear to be any unofficial kernel modules loaded that would be interfering with access to the devices.

ehidle, 2017-01-09:
AH HAH! 

Now I feel stupid. 

It turns out that I forgot to blacklist the stupid kernel-based modules on this system... and those modules were capturing the first and only the first RSP1. *facepalm*

Sorry :( All seems to be good now. 

######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

[ERROR] SoapyRemote::find() -- connect(tcp://192.168.2.4:55132) FAIL: connect(tcp://192.168.2.4:55132) [111: Connection refused]
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=160506EF00 bus=002 port=002 devAvail=1
mir_sdr_usb_GetDevices Dev1:vid=1df7 pid=2500 rev=0206 serno=B0001P0003 bus=001 port=003 devAvail=1
mir_sdr_usb_GetDevices Dev2:vid=1df7 pid=2500 rev=0206 serno=B0008P0002 bus=008 port=002 devAvail=1
Found device 0
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 36CB00B5
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 36CB00B5
  tuner = Rafael Micro R820T

Found device 1
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 99999999
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 1
  serial = 99999999
  tuner = Rafael Micro R820T

Found device 2
  driver = sdrplay
  label = SDRplay Dev0 RSP2 160506EF00

Found device 3
  driver = sdrplay
  label = SDRplay Dev1 RSP1 B0001P0003

Found device 4
  driver = sdrplay
  label = SDRplay Dev2 RSP1 B0008P0002

guruofquality, 2017-01-09:
Perhaps in the future sdrplay could use the equivalent of libusb_attach/detach_kernel_driver like the rtl to avoid manual blacklisting? http://libusb.sourceforge.net/api-1.0/group__dev.html#ga21bd343325f558987ca57e4e281a6d47

closed 2017-01-10 by guruofquality