pothosware/SoapySDR #105: compiling SoapySDR on Centos 7

magveda, 2017-01-10:
can not compile it on my Centos 7 machine: **SoapySDRUtil: error while loading shared libraries: libSoapySDR.so.0.6-dev: cannot open shared object file: No such file or directory**

```bash
sudo make install
[ 67%] Built target SoapySDR
[ 78%] Built target SoapySDRUtil
[ 82%] Built target TestFormatParser
[ 85%] Built target TestKwargsMarkup
[ 89%] Built target TestTimeConversion
[ 92%] Built target docs
[100%] Built target _SoapySDR
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Installing: /usr/local/include/SoapySDR
-- Installing: /usr/local/include/SoapySDR/Config.h
-- Installing: /usr/local/include/SoapySDR/Config.hpp
-- Installing: /usr/local/include/SoapySDR/Constants.h
-- Installing: /usr/local/include/SoapySDR/Device.h
-- Installing: /usr/local/include/SoapySDR/Device.hpp
-- Installing: /usr/local/include/SoapySDR/Errors.h
-- Installing: /usr/local/include/SoapySDR/Errors.hpp
-- Installing: /usr/local/include/SoapySDR/Formats.h
-- Installing: /usr/local/include/SoapySDR/Formats.hpp
-- Installing: /usr/local/include/SoapySDR/Logger.h
-- Installing: /usr/local/include/SoapySDR/Logger.hpp
-- Installing: /usr/local/include/SoapySDR/Modules.h
-- Installing: /usr/local/include/SoapySDR/Modules.hpp
-- Installing: /usr/local/include/SoapySDR/Registry.hpp
-- Installing: /usr/local/include/SoapySDR/Time.h
-- Installing: /usr/local/include/SoapySDR/Time.hpp
-- Installing: /usr/local/include/SoapySDR/Types.h
-- Installing: /usr/local/include/SoapySDR/Types.hpp
-- Installing: /usr/local/include/SoapySDR/Version.h
-- Installing: /usr/local/include/SoapySDR/Version.hpp
-- Installing: /usr/local/lib64/libSoapySDR.so.0.6.0
-- Installing: /usr/local/lib64/libSoapySDR.so.0.6-dev
-- Installing: /usr/local/lib64/libSoapySDR.so
-- Installing: /usr/local/lib64/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/bin/SoapySDRUtil
-- Removed runtime path from "/usr/local/bin/SoapySDRUtil"
-- Installing: /usr/local/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/lib64/python2.7/site-packages/_SoapySDR.so
-- Removed runtime path from "/usr/local/lib64/python2.7/site-packages/_SoapySDR.so"
-- Installing: /usr/local/lib64/python2.7/site-packages/SoapySDR.py
nemo@nemo-nb~/bin/SoapySDR/build $ sudo ldconfig #needed on debian systems
nemo@nemo-nb~/bin/SoapySDR/build $ SoapySDRUtil --info #test SoapySDR install
SoapySDRUtil: error while loading shared libraries: libSoapySDR.so.0.6-dev: cannot open shared object file: No such file or directory
```

guruofquality, 2017-01-10:
/usr/local/lib64 is probably not a library path that is searched by default. Try this:

```
export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH
SoapySDRUtil --info
```

So the library path can be set in the bash profile like that, or more
permanently the path can be added for ldconfig:
https://www.cyberciti.biz/faq/linux-setting-changing-library-path/

If it also helps, another option is to change the CMAKE_INSTALL_PREFIX
and LIB_SUFFIX passed to cmake to something the system is already
looking for. Whatever works best~ 

* Reference https://groups.google.com/d/msg/pothos-users/D0supuq4X30/567CfdPQEAAJ

magveda, 2017-01-10:
You are right!

To make it permanent I created file `sudo vim /etc/ld.so.conf.d/usrlocal.conf ` with contents

```
/usr/local/lib/
/usr/local/lib64/
```

then `sudo ldconfig` and everything is good.

Centos Rocks!

closed 2017-01-10 by magveda