pothosware/SoapySDR #107: Cannot run Python version, OS X

dmitryelj, 2017-01-24:
Hi all.

I'm trying to test Python version: 

with **python SimpleSiggen.py**
Getting this error:
_bash-3.2$ python SimpleSiggen.py 
Traceback (most recent call last):
  File "SimpleSiggen.py", line 5, in <module>
    import SoapySDR
  File "/usr/local/lib/python2.7/site-packages/SoapySDR.py", line 28, in <module>
    _SoapySDR = swig_import_helper()
  File "/usr/local/lib/python2.7/site-packages/SoapySDR.py", line 24, in swig_import_helper
    _mod = imp.load_module('_SoapySDR', fp, pathname, description)
ImportError: dlopen(/usr/local/lib/python2.7/site-packages/_SoapySDR.so, 2): Library not loaded: libSoapySDR.0.5-1.dylib
  Referenced from: /usr/local/lib/python2.7/site-packages/_SoapySDR.so
  Reason: image not found_

In the folder /usr/local/lib/python2.7/site-packages/ there is a file SoapySDR.py, that has pretty old date.
I tried to build a new one:
swig -python SoapySDR.i

Got a error:
/opt/local/share/swig/3.0.10/python/std_complex.i:13: Error: Syntax error in input(1)

My system: OS X 10.11.6
Thanks.

guruofquality, 2017-01-24:
How did you install SoapySDR? One issue I ran into OSX is that python detection was sometimes difficult because there was an installed python and one provided by another tool. Like you see above two different pythons were detected. To work around this, and you can see it in the brew recipe, I added a flag to cmake (`-DUSE_PYTHON_CONFIG=ON`) that finds the python-config executable, and asks it for library and includes to get a consistent answer.

* Also I have this home brew tap if that helps: https://github.com/pothosware/homebrew-pothos/wiki

dmitryelj, 2017-01-25:
Hi,

Thanks for the answer. 
I'm trying to build Soapy myself.

**cmake .. -DUSE_PYTHON_CONFIG=ON** helps, the build was success. In the installer output I see:
-- Installing: /usr/local/lib/python2.7/site-packages/_SoapySDR.so
-- Installing: /usr/local/lib/python2.7/site-packages/SoapySDR.py
-- Installing: /usr/local/lib/python3.5/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python3.5/site-packages/SoapySDR.py

But when I try the example, I get **ImportError: No module named SoapySDR**.
Finally, I put SoapySDR.py and _SoapySDR.so in the same folder with the sample, its ugly but seems working (didn't check with real receiver yet).

Its always messy on MacOS, because there are 2 python interpreters on it, so a bit dll/library hell.

PS: From your last link:
brew install soapysdr - returns "No available formula with the name "soapysdr"

guruofquality, 2017-01-25:
> But when I try the example, I get ImportError: No module named SoapySDR.
Finally, I put SoapySDR.py and _SoapySDR.so in the same folder with the sample, its ugly but seems working (didn't check with real receiver yet).

I suppose /usr/local/lib/python2.7/site-packages isnt in the default python search path. I recommend setting the PYTHONPATH

> PS: From your last link:
brew install soapysdr - returns "No available formula with the name "soapysdr"

Did you add the tap first?

guruofquality, 2017-02-02:
Resolved I think

closed 2017-02-02 by guruofquality