pothosware/SoapySDR #109: latest soapy commits for gr-osmosdr

starvald, 2017-02-05:
Sorry if this isn't the place but i can't find an issues tab in gr-osmosdr.

re: 2 Commits on Feb 2, 2017
recompiled gr-osmosdr as i noticed the latest commits but it's only half working.

the waterfall is showing a signal but the panadapter is flatlined, tried my airspy and sdrplay pro2.

[image]

guruofquality, 2017-02-05:
So the only major change in gr-osmosdr is that devices supported through SoapySDR will show up in the device list in GQRX. Provided that soapysdr was installed, and modules for like airspy and sdrplay are available.

So I doubt that there is any reversion, but are you perhaps comparing SoapyAirspy (which is in the screenshot) to the behaviour of the [lib/airspy](https://github.com/osmocom/gr-osmosdr/tree/master/lib/airspy) support  in gr-osmosdr? If so, I want to make sure that nothing is broken in gr-osmosdr. Can you still select the original airspy and sdr play that are directly an option in gr-osmosdr compile options?

And if there is some difference for the controls, maybe we should fix SoapyAirspy or SoapySDRPlay to match. I don't know what control could break the panadapter, so I'm not sure how this could be affected by different hardware modules. @csete sorry to bother, but any idea about the screenshot above?

csete, 2017-02-05:
I think it's a gqrx problem that can happen regardless of the driver backend. Try resetting the configuration using "gqrx -r" (nb: you will loose settings but I have no other solution at the moment)

starvald, 2017-02-05:
Sorted, i needed to reset the gqrx config as you suggested.

guruofquality, 2017-02-05:
ok looks like everything is ok. thanks!

closed 2017-02-05 by guruofquality