pothosware/SoapySDR #112: API addtion to set digital filter taps

guruofquality, 2017-02-21:
We have had to set taps through the arbitrary settings API using CSV format, and setting taps as a vector of complex numbers would be an ideal API addition for several devices that support configurable FIR filters, either in the FPGA or RFIC.

For example:
```
void setFilterTaps(int direction, size_t channel,
    const std::vector<std::complex<double>> &taps)
```

Consider an API call to query the support for this call and the maximum number of filter coefficients.

OhSoGood, 2019-08-14:
@guruofquality : Josh, any news on this? I would have to change the FIR of my SDRs (a bladerf here) that I'm using with Soapy, and wonder how to do it. Today, is it possible to mix soapy with direct call to the SDR dll, namely set_filter in bladerf.dll (how can we get a pointer to the underlying sdr object?)?
Thanks

guruofquality, 2019-08-14:
In the interim since creating this issue there was a is a call added `writeRegisters` that can handle writing a contiguous block of numbers. I have used it to write taps, block rams, coefficients and things like that. Its a bit more general and has done the job, so I guess the more specific API of filter taps kind of got ignored.

OhSoGood, 2019-08-15:
I had a look to the code of the bladerf lib, and I am not sure we can use the register to set the FIR filters. Maybe I am wrong (I'm not used to directly handling low-level registers)? Could you have a look in the blade code? 

I think the entry point to look at is the calls in bladerf2.c to rfic->set_filter (ex in methods [bladerf_set_rfic_tx_fir](https://github.com/Nuand/bladeRF/blob/055c7dfb29e4e93f3ccef6f8627f693ef1664eec/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c#L3135) or [bladerf2_set_sample_rate](https://github.com/Nuand/bladeRF/blob/055c7dfb29e4e93f3ccef6f8627f693ef1664eec/host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c#L1056):  in bladerf2.c)
Thank you!

guruofquality, 2019-08-15:
Ill put it this way. If there is an API to set filter taps, writeRegisters would be pretty helpful. If there is an API to set particular filter bank, then writeSetting will do the trick.

In fact, there should be one already for the xb200 to support bladerf_xb200_set_filterbank. But I don't see an API exposed for filter taps, maybe no one is actually doing that. So maybe that something you can request from nuand? https://github.com/Nuand/bladeRF/issues

And if something is missing specific to blade rf, please open an issue or make a PR with:
https://github.com/pothosware/SoapyBladeRF