pothosware/SoapySDR #114: Compiling installs the lib in a strange path

bogdanr, 2017-02-26:
I am compiling it like this:
`cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -DLIB_INSTALL_DIR:PATH=lib64 ..`

but when I run `make install`, the libs, such as (libSoapySDR.so.0.6-dev) end up in `/urs/lib` instead of `/usr/lib64`

Moving the libs manually to `/usr/lib64` makes thing work.

guruofquality, 2017-02-26:
Use `-DLIB_SUFFIX=64` (see https://github.com/pothosware/SoapySDR/blob/master/lib/CMakeLists.txt#L80)

We try to guess this, but its not perfect: https://github.com/pothosware/SoapySDR/blob/master/cmake/SoapySDRConfig.cmake#L26

Just curious, but what OS/linux flavour are you installing on?

bogdanr, 2017-02-27:
I am installing on [NimbleX](http://nimblex.net/) which is somewhat close to Slackware-current as the build environment is concerned.

closed 2017-03-16 by guruofquality