pothosware/SoapySDR #116: Device::make table is keyed by exact args list, not device identity

kevinmehall, 2017-03-05:
Many different args lists will open the same device. If you call `Device::make` with an args list it hasn't seen before, it will try to instantiate a new `Device`, even when a cached `Device` for the same hardware may already exist.

For drivers that also do their own internal device caching like `lime`, this can cause bigger problems than simply failing to open an already-claimed device. As an example, in Python:
```
import SoapySDR
dev1 = SoapySDR.Device('driver=lime')
print "==========="
dev2 = SoapySDR.Device('driver=lime,foo=bar')
print hex(dev1.this), hex(dev2.this)
```
When creating `dev2`, the already-opened device is re-initialized. `dev1` and `dev2` pointers differ, thus they are separate `SoapyLMS7` instances, each with their own `_accessMutex`, but both pointing to the same `lime::IConnection`.

One possibility is to use the result of `Device::enumerate` as the key to the table. This would require drivers to always return the same args list for the same device, which seems like a better assumption than the user always entering the device string in the same way. However, this is not always the case, for instance `rtlsdr` violates this with its `available` arg.

kevinmehall, 2017-03-05:
It seems like `getDriverKey` and `getHardwareKey` have the ideal semantics for use in enumeration, but are only available on already-opened devices. Perhaps there needs to be some concept of a "canonical" args list that usually only consists of a driver and a serial number or device path.

guruofquality, 2017-03-05:
Good catch, it was supposed to use the result of enumerate() so this would not happen. This is a pending fix if you want to give it a try: https://github.com/pothosware/SoapySDR/tree/fix_device_table_keying

closed 2017-03-07 by guruofquality