pothosware/SoapySDR #118: soapy_power  ImportError: No module named concurrent.futures

bogdanr, 2017-03-10:
I am trying to run `soapy_power ` but I get an error I don't understand why it happens:
```
nimblex:~/SoapySDR/build# soapy_power 
Traceback (most recent call last):
  File "/usr/bin/soapy_power", line 9, in <module>
    load_entry_point('soapy-power==1.0.0', 'console_scripts', 'soapy_power')()
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib64/python2.7/site-packages/soapypower/__main__.py", line 6, in <module>
    from soapypower import power
  File "/usr/lib64/python2.7/site-packages/soapypower/power.py", line 8, in <module>
    from soapypower import psd, writer
  File "/usr/lib64/python2.7/site-packages/soapypower/psd.py", line 3, in <module>
    import math, logging, concurrent.futures
ImportError: No module named concurrent.futures
```
```
nimblex:~/SoapySDR/build# python3 -c "import concurrent.futures"
nimblex:~/SoapySDR/build# echo $?
0
```

guruofquality, 2017-03-10:
I dont recognize the soapypower module, but its being run in a python2 interpreter, but its code seems to rely on python3 only module `concurrent`

xmikos, 2017-03-11:
[soapy_power](https://github.com/xmikos/soapy_power) is my project (written mainly as new universal backed for [QSpectrumAnalyzer)](https://github.com/xmikos/qspectrumanalyzer). So any bugreports please send there.

But this is not a bug, it seems you have somehow installed it with Python 2 instead of Python 3.

closed 2017-03-13 by bogdanr