pothosware/SoapySDR #119: soapysdr.rb: Python.h file not found

lool, 2017-03-15:
Hi,

`brew install soapysdr --HEAD`
fails to build for me even with `--with-python3`; I have both python and python3 formulas installed.

```
==> Checking out branch master
==> cmake .. -DUSE_PYTHON_CONFIG=ON -DENABLE_PYTHON3=OFF -DSOAPY_SDR_ROOT='/usr/local' -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_
==> make install
[...]
[ 96%] Building CXX object python/CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o
cd /tmp/soapysdr-20170315-46983-hzbmec/build/python && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -D_SoapySDR_EXPORTS -I/tmp/soapysdr-20170315-46983-hzbmec/include -I/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7  -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -std=gnu++11 -o CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o -c /tmp/soapysdr-20170315-46983-hzbmec/build/python/SoapySDRPYTHON_wrap.cxx
/tmp/soapysdr-20170315-46983-hzbmec/build/python/SoapySDRPYTHON_wrap.cxx:174:11: fatal error: 'Python.h' file not found
# include <Python.h>
```

`python-config --include` does report a bunch of include pathes where I do see a Python.h, and this works:
```
echo '#include "Python.h"' | cpp `python3-config --include`
```

Oddly enough, if I build with --debug and open a shell, then copy-paste the last line of the build supposedly triggering the failure:
```
cd /tmp/soapysdr-20170315-53162-1lhhyv3/build/python && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -D_SoapySDR_EXPORTS -I/tmp/soapysdr-20170315-53162-1lhhyv3/include -I/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7  -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -std=gnu++11 -o CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o -c /tmp/soapysdr-20170315-53162-1lhhyv3/build/python/SoapySDRPYTHON_wrap.cxx
```
...it passes, in fact `make install` passes. So I'm a bit puzzled as to why the brew install build fails. Any idea?

Thanks,
- Loïc Minier

guruofquality, 2017-03-15:
Its building both python2 and python3 bindings, so the issue seems to be with python2. Can you make with verbose? I think its --verbose option passed to brew. The configure output from cmake prints which flags and libs are selected for python. And I know that brew python can sometimes get confused because osx comes with python2 as well.

lool, 2017-03-15:
With `brew install soapysdr --HEAD --verbose`:
```
[ 93%] Linking CXX executable SoapySDRUtil
cd /tmp/soapysdr-20170315-57702-8oj7f6/build/apps && /usr/local/Cellar/cmake/3.7.2/bin/cmake -E cmake_link_script CMakeFiles/SoapySDRUtil.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -Wl,-search_paths_first -Wl,-headerpad_max_install_names  CMakeFiles/SoapySDRUtil.dir/SoapySDRUtil.cpp.o CMakeFiles/SoapySDRUtil.dir/SoapySDRProbe.cpp.o CMakeFiles/SoapySDRUtil.dir/SoapyRateTest.cpp.o  -o SoapySDRUtil -Wl,-rpath,/tmp/soapysdr-20170315-57702-8oj7f6/build/lib  ../lib/libSoapySDR.0.6.0.dylib 
[ 93%] Built target SoapySDRUtil
cd /tmp/soapysdr-20170315-57702-8oj7f6/build && /usr/local/Cellar/cmake/3.7.2/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/soapysdr-20170315-57702-8oj7f6 /tmp/soapysdr-20170315-57702-8oj7f6/python /tmp/soapysdr-20170315-57702-8oj7f6/build /tmp/soapysdr-20170315-57702-8oj7f6/build/python /tmp/soapysdr-20170315-57702-8oj7f6/build/python/CMakeFiles/_SoapySDR.dir/DependInfo.cmake --color=
Scanning dependencies of target _SoapySDR
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f python/CMakeFiles/_SoapySDR.dir/build.make python/CMakeFiles/_SoapySDR.dir/build
[ 96%] Building CXX object python/CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o
cd /tmp/soapysdr-20170315-57702-8oj7f6/build/python && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -D_SoapySDR_EXPORTS -I/tmp/soapysdr-20170315-57702-8oj7f6/include -I/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7  -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -std=gnu++11 -o CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o -c /tmp/soapysdr-20170315-57702-8oj7f6/build/python/SoapySDRPYTHON_wrap.cxx
/tmp/soapysdr-20170315-57702-8oj7f6/build/python/SoapySDRPYTHON_wrap.cxx:174:11: fatal error: 'Python.h' file not found
# include <Python.h>
          ^
```

lool, 2017-03-15:
So I've dumped environment before running the clang++ call and bisected the problematic env var to be:
HOMEBREW_CCCFG="Osa"
if I remove it, the build passes; this seems to be a feature to do something clever about the build flags

guruofquality, 2017-03-15:
I was going to say that I was actually looking for the cmake output at the beginning, not the error at the end. But it looks like you are on to something strange about brew... Literally no idea on my end, but it looks like the SoapySDR build is coming up with the right flags for python. Let us know if you figure it out or if the recipe can be changed in some way to support that var.

lool, 2017-03-15:
Hmm sorry, this should be against homebrew-pothos

lool, 2017-03-15:
So adding the python build-dep fixes the build for me! See https://github.com/pothosware/homebrew-pothos/pull/7

dpruessner, 2017-03-16:
Encountered this problem with "Python.h" file not found; confirming lool's fix allowed compile build to complete on OS X 10.11.6.

guruofquality, 2017-03-16:
I merged the recommends python change. I think we are supposed to use the system python. But since thats not compiling... Cant argue with that.

closed 2017-03-16 by guruofquality