pothosware/SoapySDR #122: Provide minimal SoapySDR installer for Windows (preferably also 32-bit version)

xmikos, 2017-03-22:
Right now the only option for Windows users to get SoapySDR libraries (and basic `SoapySDRUtil` executable) is [PothosSDR installer bundle](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D), which is rather large (more than 500MB after installation), contains many applications (e.g. GNU Radio, GQRX, CubicSDR, Pothos GUI, Zadig, etc.) and is only available for 64-bit Windows.

It would be great if you could provide minimal installer containing only SoapySDR, SoapySDR device drivers (all drivers and theirs required libraries are only about 20MB together), Python SoapySDR bindings and `SoapySDRUtil` executable. And it would be even better if you could also provide installer for 32-bit Windows.

Alternatively SoapySDR Python bindings could be separated and provided on [PyPI](https://pypi.python.org/pypi) which is standard for Python packages nowadays (even large bindings like PyQt5 are now available in PyPI).

guruofquality, 2017-05-03:
The actual installer is ~80MB, so I think the size issue on install could be handled with NSIS components

guruofquality, 2017-05-03:
This issue was moved to pothosware/PothosSDR#37

closed 2017-05-03 by guruofquality