pothosware/SoapySDR #123: Gqrx crash due to setFrequency(CORR)

romeojulietthotel, 2017-04-02:
Here's the error that is returned when Gqrx crashes.

`what():  SoapyLMS7::setFrequency(CORR) unknown name
`

It looks like gr-osmosdr is the only part of the chain that knows about the "CORR" name. SoapyLMS7 only knows about "BB" and "RF".

I traced this to gr-osmosdr but maybe it is actually something missing from the implementation in SoapySDR or SoapyLMS7?

I see this:

https://github.com/pothosware/SoapySDR/issues/84

and this

https://github.com/osmocom/gr-osmosdr/commit/6d6a483cfbe7ecae6d0858ef862d44405efb6e6f#diff-e0bd62e7d68642e3309d792caa736bd5

and this

https://github.com/csete/gqrx/issues/496

guruofquality, 2017-04-02:
Ahh, the set_freq_corr() call is not universally implemented in gr-osmosdr and we dont have anyway to check if it exists. I think the appropriate fix is to check within the osmo/soapy wrapper for this value, or just nicely ignore it.

* Can you try this branch: http://git.osmocom.org/gr-osmosdr/log/?h=soapy_support
* This commit http://git.osmocom.org/gr-osmosdr/commit/?h=soapy_support&id=c8af6b1735950a5feb268b44bc34b1ccd3c07392

* and eventually, its probably better dealt with with an API for corrections like this: https://github.com/pothosware/SoapySDR/issues/84

romeojulietthotel, 2017-04-03:
Yes that work-around works to avoid a crash. Thanks.

guruofquality, 2017-04-20:
Fixed on gr-osmosdr master, also using the new freq correction api when applicable.

closed 2017-04-20 by guruofquality