pothosware/SoapySDR #127: listClockSources and getClockSource return empty for LimeSDR widget

kb1vc, 2017-05-07:
listClockSources returns an empty list. 
getClockSource returns an empty string. 

SoapySDR version shown below. 
 
```
SoapySDR version API [0.6.0] ABI [0.6-dev] Lib [0.6.0-g8f459a47]
Device 0
	Attr [addr] = [1d50:6108]
	Attr [driver] = [lime]
	Attr [label] = [LimeSDR-USB [USB 2.0] 9060B00461821]
	Attr [media] = [USB 2.0]
	Attr [module] = [STREAM]
	Attr [name] = [LimeSDR-USB]
	Attr [serial] = [0009060B00461821]
```
All other methods I've used seem to work just fine.

guruofquality, 2017-05-10:
Well, thats true that listClockSources returns an empty list. At present its actually not implemented. There is a reference input for LimeSDR, I think its automatic switching, so there is no software hook to implement here (I think). Let me know if otherwise. Or should listClockSources just return something so it looks more sensible in a GUI. If so, I'm happy to take a suggestion.

guruofquality, 2017-09-29:
If there is a specific feature or use case missing for LimeSDR, please open an issue or PR on LimeSuite (thats where the implementation is).

closed 2017-09-29 by guruofquality