pothosware/SoapySDR #129: Python3 install dir is incorrect for Linux Mint (and possibly Ubuntu)

vchekan, 2017-05-27:
OS: Linux Mint 18.1 Serena
SoapySDR: current git: 74f890c

After installing from sources, an attempt to use it fails:

```
Traceback (most recent call last):
  File "/home/vadim/.local/bin/soapy_power", line 7, in <module>
    from soapypower.__main__ import main
  File "/home/vadim/.local/lib/python3.5/site-packages/soapypower/__main__.py", line 5, in <module>
    import simplesoapy
  File "/home/vadim/.local/lib/python3.5/site-packages/simplesoapy.py", line 5, in <module>
    import SoapySDR
ImportError: No module named 'SoapySDR'
```

SoapySDR module is installed into '/usr/local/lib/python3/dist-packages' but this path is not in sys.path:

```
vadim@desktop /usr/local/lib/python3/dist-packages $ python3
Python 3.5.2 (default, Nov 17 2016, 17:05:23) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys; sys.path
['', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/vadim/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages']
>>> 
```

Notice, that '/usr/lib/python3/dist-packages' does exist in sys.path, and this is what distutils.sysconfig.get_python_lib() would return:
https://github.com/pothosware/SoapySDR/blob/5e823fc114898bfe1b4ee8b874ce825665938c8e/python3/CMakeLists.txt#L29

But adding ${prefix} (/usr/local) to this folder is incorrect, because "lib/python3" does exist for "/usr" but not for "/usr/local". 
Thus CMake will install SoapySDR into folder which is not in sys.path and it won't be found by python.

guruofquality, 2017-05-27:
I think this is ok, and basically the way to do it. The system python only searches system paths by default. /usr/local happens to be the default for compiled software, but you have to set stuff like LD_LIBRARY_PATH and PYTHONPATH for the python libs to get it work. Of course if this was a binary package/installer it would install to /usr

Anyway, a few options:
* use as is and set PYTHONPATH to /usr/local/lib/python3/dist-packages in .bashrc/profile
* custom python install dir passed to cmake with -DPYTHON_INSTALL_DIR=
* or move entire install prefix to the /usr with -DCMAKE_INSTALL_PREFIX

vchekan, 2017-05-29:
The problem is, other pyton packaged do manage to install themselves into locations which are in the sys.path whereas SoapySDR does not. 

PYTHONPATH would work, but the question is, why create a problem to find workaraund for it? 

/usr prefix is undesirable because it is much more difficult to clean up manually.

Given that sys.path does contains entries for both, /usr and /usr/local, I still consider it a bug to install module under /usr/local/lib/python3 which is not in python path while ignoring /usr/local/python3.5 which is present in python path.

Is there any reason why setup.py installation method is not used? I could try to implement it if there is no objections against it.

guruofquality, 2017-07-12:
> The problem is, other pyton packaged do manage to install themselves into locations which are in the sys.path whereas SoapySDR does not.

Its really the same issue with the shared library, /usr/local/lib and similar is generally not in the LD_LIBRARY_PATH and /usr/local/* for python is generally not in sys.path. And thats going to depend on the platform specifics as well.

The general paradigm when having source installs is to make sure that everything goes into the prefix, and let the user set the environment accordingly. You setup the environment once and there is usually a lot of other packages installed along side in this install prefix.

I get the convenience of having all of the python modules and libraries found automatically for a default source install. But its hard to know ahead of time what those paths are, or even if the particular OS will support that (searing something outside of /usr/ by default). So we make a best effort, and give lots of variables to modify the installation paths.

> Given that sys.path does contains entries for both, /usr and /usr/local, I still consider it a bug to install module under /usr/local/lib/python3 which is not in python path while ignoring /usr/local/python3.5 which is present in python path.

It may well be a bug. We ask python for the directory like so:
`from distutils.sysconfig import get_python_lib; print(get_python_lib(plat_specific=True, prefix=''))`

Cmake cant know the directory naming convention on any given platform, so we just ask python. Maybe there is a better or more correct way. In any case, if it helps, this can be overridden with `-DPYTHON_INSTALL_DIR` if need be.

> Is there any reason why setup.py installation method is not used? I could try to implement it if there is no objections against it.
https://docs.python.org/3/distutils/setupscript.html

SoapySDR has development files among other things, so I suppose that a setup.py use case would be for pure python usage. So I assume setup.py can invoke cmake and setup things like PYTHON_INSTALL_DIR and CMAKE_INSTALL_PREFIX for whatever is appropriate for setup.py

pztrick, 2018-03-17:
I tend to agree with OP. The default install path is not in python3's default PYTHONPATH.

However, I am encountering an issue importing when I change the path using the `-DPYTHON_INSTALL_DIR` flag.

```
cmake .. -DPYTHON_INSTALL_DIR=/usr/lib/python3/dist-packages
# -- Installing: /usr/lib/python3/dist-packages/_SoapySDR.so
# -- Installing: /usr/lib/python3/dist-packages/SoapySDR.py
# -- Installing: /usr/local/lib/python3/dist-packages/_SoapySDR.so
# -- Installing: /usr/local/lib/python3/dist-packages/SoapySDR.py
python3 -c 'import SoapySDR'
```

This produces this `ImportError`:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/SoapySDR.py", line 28, in <module>
    _SoapySDR = swig_import_helper()
  File "/usr/lib/python3/dist-packages/SoapySDR.py", line 24, in swig_import_helper
    _mod = imp.load_module('_SoapySDR', fp, pathname, description)
  File "/usr/lib/python3.5/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: dynamic module does not define module export function (PyInit__SoapySDR)
```

---

On the other hand, using `cmake` without the Python path flag and adding sym-links works great:

```
sudo ln -svf /usr/local/lib/python3/dist-packages/SoapySDR.py /usr/lib/python3/dist-packages/
sudo ln -svf /usr/local/lib/python3/dist-packages/_SoapySDR.so /usr/lib/python3/dist-packages/
```

Not clear to me why this should work but not the first approach.

Anyway, I'm working on adding a small PyPi package to install into `virtualenv` environments and will report back if I succeed...

pztrick, 2018-03-17:
I uploaded a package named `vext.SoapySRD` to PyPI that can be used to install `SoapySRD` into virtualenvs:

https://github.com/pztrick/vext.SoapySDR

```
pip install vext.SoapySDR
```

You must build `SoapySDR` on your system first.

cswiger, 2018-03-17:
I just stumbled on this and it looks like Ubuntu 16.04 for some reason has both /usr/local/lib/python3 and /usr/local/lib/python3.5  -  python3.5 is in the path ok (and has SimpleSoapy, SimpleSpectral and soapy_power) , and have to add PYTHONPATH to python3 to get SoapySDR working.  

Maybe due to SoapySDR/debian/python3-soapysdr.install  has wildcard usr/lib/python3*/  that matches first - but that's not /usr/local/lib/   hmmmm.    

Ubuntu normally finds things in /usr/local/  ok after build and ldconfig  w/o need for LD_LIBRARY_PATH etc.

guruofquality, 2018-03-18:
@pztrick this error usually comes up when python3 imports a python2 modules or vice versa. SoapySDR can build for both at the same time, take a closer look at the cmake output to see which versions of python it build with and/or try PYTHON3_INSTALL_DIR

```
> cmake .. -DPYTHON_INSTALL_DIR=/usr/lib/python3/dist-packages
# -- Installing: /usr/lib/python3/dist-packages/_SoapySDR.so
# -- Installing: /usr/lib/python3/dist-packages/SoapySDR.py
# -- Installing: /usr/local/lib/python3/dist-packages/_SoapySDR.so
# -- Installing: /usr/local/lib/python3/dist-packages/SoapySDR.py
python3 -c 'import SoapySDR'

This produces this ImportError:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/SoapySDR.py", line 28, in <module>
    _SoapySDR = swig_import_helper()
  File "/usr/lib/python3/dist-packages/SoapySDR.py", line 24, in swig_import_helper
    _mod = imp.load_module('_SoapySDR', fp, pathname, description)
  File "/usr/lib/python3.5/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: dynamic module does not define module export function (PyInit__SoapySDR)
```

guruofquality, 2018-03-18:
@cswiger I don't think ubuntu/debian has any default python search paths in /usr/local. When the SoapySDR build is configured for installation into /usr (like when making debian packages), it will install into default python search paths for python2 and 3 respectively. You can see this in the myriadrf/drivers ppa and the official ubuntu packages for soapysdr

When building from source, SoapySDR merely uses the /usr/local version of these paths by default. Thats the general convention for many of the sdr/python/gnuradio style applications. The upside is that its basically very file system hierarchy friendly, and allows multiple installations into different prefixes that are usable with a simple bash source script thats sets variables like PATH, LD_LIBRARY, PYTHONPATH, etc...

And for anyone else listening: On the other hand, yes I agree its annoying because people building from source will probably have to set the PYTHONPATH in their bash profile or something to that effect. Also, I dont recommend doing source installs into /usr/, thats typically reserved for package managers. Of course be my guest its not my machine :-)

And anyone installing python modules by some other means setup.py and the like is totally welcome to select a custom PYTHON_INSTALL_DIR and PYTHON3_INSTALL_DIR for whatever configuration is expected, or move the files around as needed. But if I change the default behaviour, its going to break community conventions and unexpectin redhat/debian/etc packagers and I will just have another similar issue open asking me to change it back :-P

> Ubuntu normally finds things in /usr/local/ ok after build and ldconfig w/o need for LD_LIBRARY_PATH etc.

Yeah, thats a debian quirk. If the library path isnt set, the cache has to be updated for it to see any new libraries in the default search paths. Every deb that installs .so libs actually has a postinst just to run ldconfig.

guruofquality, 2018-09-11:
Maybe this will help: https://github.com/pothosware/SoapySDR/commit/a67a970ac47f4c2d6c10810053dd5713b1e6228d

I still think something is strange with how the python distutils and or install is configured or why its different from its python2 counterpart. Is it a bug in c python, is it a bug in how debian/ubuntu configures the package.

closed 2018-09-11 by guruofquality