pothosware/SoapySDR #130: Unable to install Python3 module without Python2

blshkv, 2017-07-12:
It seems a logic flaw in the current implementation:
```
if (ENABLE_PYTHON)
 ...
   if(BUILD_PYTHON3)
   endif(BUILD_PYTHON3)
endif (ENABLE_PYTHON)

Shouldn't it be independent check?

it looks like you will need an additional flag BUILD_PYTHON2 to make it clear.

guruofquality, 2017-07-12:
The python build uses the standard cmake findpython and libs stuff so it can handle both python2 and python3. So if you want to build only for python3 just use ENABLE_PYTHON and the non specific python variables.

The python3 only lets some builds do both python2 and python3. Its supposed to be helpful for some development and build environments that have both supported.

blshkv, 2017-07-12:
That basically means that i cant command to build python3 only if i have both versions installed.

But that's exactly that im asking to help with. Im creating an ebuild for Gentoo where there are two variables python2 and python3 can be set. Python2 module should be disabled if its variable is not set

guruofquality, 2017-07-12:
> That basically means that i cant command to build python3 only if i have both versions installed.

You can build python3 only. The cmake python stuff isnt very smart but if you tell it to use python config it should find all of the relevant build dependencies. if you have a python3-config or similar variable do this:

```
cmake ~/src/SoapySDR -DPYTHON_EXECUTABLE=/usr/bin/python3 -DUSE_PYTHON_CONFIG=ON
```

blshkv, 2017-07-12:
well, yeah. There is an easier hack too, just delete ```add_subdirectory(python)``` line.

but can you fix the main CMakLists file so it would an "official" way on the configure stage from the top directory?

guruofquality, 2017-07-14:
ok, I understand, its not that you can't do this, its that cmake isnt doing a good job of automatically finding python3 when its a python3 only system. Enabling the python3 directory when the python/CMakeLists.txt fails to find anything might be a good idea. I put it on master, give it a shot: https://github.com/pothosware/SoapySDR/commit/9781ae5674b2ebfe25bde45ad4e7dc128542d41f

> well, yeah. There is an easier hack too, just delete add_subdirectory(python) line.

That shouldnt work, the python3/CMakeLists.txt directory is still guarded by BUILD_PYTHON3. Basically its setup so if the first python directory gets setup for python v3 we dont need to enable python3/CMakeLists.txt But hopefully the change above basically enables this.

blshkv, 2017-07-14:
hm... I haven't try this  but I'm not sure what it is what I'm asking.

I have both python2 and python3 installed, but I would like to configure soapysdr to support python3 only

ENABLE_PYTHON=YES will activate both
BUILD_PYTHON3=YES will force to build python3 module.

But how to disable building python2 module without disabling python3?

guruofquality, 2017-07-17:
The python/CMakeLists.txt uses the official FindPythonInterp.cmake and FindPythonLibs.cmake modules and will support any version of python. To build for python3 only, the recommended and the official way is to configure the variables in python/CMakeLists.txt to build against the desired version of python.

The python3/CMakeLists.txt exists specially as a convenience for installing on dual python2/python3 systems like ubuntu. Because of this intended used, python3/CMakeLists.txt gets disabled when the python/CMakeLists.txt is configured to already build for python3.

Now these FindPythonInterp.cmake and FindPythonLibs.cmake are bad sometimes its a pain to configure. So the most reiable way to configure for a specific version of python is to pass in the executable and to `USE_PYTHON_CONFIG=ON` which uses the development files recommended by the python executable. This works especially well on platforms with too many python installations like travisci or osx.
```
cmake ~/src/SoapySDR -DPYTHON_EXECUTABLE=path/to/pythonX -DUSE_PYTHON_CONFIG=ON
```

I think what you want is to enable the python3/CMakeLists.txt and to disable the python/CMakeLists.txt because python3/CMakeLists.txt is very smart at automatically finding python3 specifically and python/CMakeLists.txt usually finds python2 by default. I didnt think to use it this way, it just wasnt the original thought or intention but I think that I could change the build such that this would work:

```
cmake ~/src/SoapySDR -DENABLE_PYTHON=OFF -DENABLE_PYTHON3=ON #TODO
```

So thats a possibility too with some minor tweaks, but its special argument either way. I think i will add it, and there will just be more than one way to build for python3 only.

blshkv, 2017-07-17:
Thank you for understanding. As a user, I don't want to know about subfolders such as python(why not python2?)/python3 . The typical way is configure the package from the top level directory (https://github.com/pothosware/SoapySDR/blob/master/CMakeLists.txt) using available variables.

So yes, it would be good two have just two simple variables ```ENABLE_PYTHON``` and ```ENABLE_PYTHON3```

guruofquality, 2017-07-17:
>  I don't want to know about subfolders such as python(why not python2?)/python3

Its just a matter of project history. "python" was there first, and it handles all versions of python, not specifically python 2*. "python3" was added later in the project for special cases.

> The typical way is configure the package from the top level directory using available variables

Larger projects will use directories. We could flatten everything into one CMakeLists.txt, but the usage of variables would be the same

blshkv, 2017-07-17:
we have some kind of misunderstanding going on because I'm lazy. My apologies.
I totally understand the historical issue and the flatten structure would be a mess.

I'm just asking to provide high level (read: human/"user" readable) variables in the top CMakeLists.txt file.
It, in its turn, should call whatever is required (such as long ```-DPYTHON_EXECUTABLE=/usr/bin/python3 -DUSE_PYTHON_CONFIG=ON``` line which make sense to developers) to activate these commands.

blshkv, 2017-07-17:
I can see the following two options:
```
if (ENABLE_PYTHON)
  if (BUILD_PYTHON2)
    add_subdirectory(python)
..
  if (BUILD_PYTHON3)
    add_subdirectory(python3)
```
or even simpler (as you suggested):
```
if (ENABLE_PYTHON)
    add_subdirectory(python)
..
if (ENABLE_PYTHON3)
  add_subdirectory(python3)
```

guruofquality, 2017-07-17:
So I added the option `-DENABLE_PYTHON=OFF`, and the python3 directory will still operate. Both build options operate through cmake_dependent_option() and should show up with the standard checkboxes and all that when you use cmake-gui

This does change the effect of `ENABLE_PYTHON` because it no longer disables all python builds totally since `ENABLE_PYTHON3` can still be enabled independently. Because some may be using ENABLE_PYTHON as a way to disable all python support entirely.

closed 2017-07-17 by blshkv