pothosware/SoapySDR #132: install FILES given no DESTINATION!

bookdude, 2017-07-27:
Fails when installing as part of gr-gsm pybomb.

Installs without issue stand alone via clone \ cmake on the same system.

root@pi-gsm-01:~/git/SoapySDR/build# pybombs install gr-gsm
PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-gsm
   |
   \- gr-osmosdr
      |
      +- airspy
      |
      +- hackrf
      |
      \- soapysdr
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: soapysdr
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/soapysdr/build
Configuring: (100%) [======================================================================================================================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 2.0.12
-- PYTHONINTERP_FOUND: TRUE - 2.7.12
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.12
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7
-- PYTHON_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpython2.7.so
-- CMAKE_SWIG_FLAGS=-c++;-threads;-DSIZE_T_IS_UNSIGNED_INT
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 2.0.12
-- Could NOT find Python3InterpDbg (missing:  PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.5/distutils/sysconfig.py", line 14, in <module>
    import re
  File "/usr/lib/python3.5/re.py", line 335, in <module>
    import copyreg
  File "/usr/local/lib/python2.7/dist-packages/copyreg/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 53, in apport_excepthook
    if not enabled():
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
    import re
  File "/usr/lib/python3.5/re.py", line 335, in <module>
    import copyreg
  File "/usr/local/lib/python2.7/dist-packages/copyreg/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.

Original exception was:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.5/distutils/sysconfig.py", line 14, in <module>
    import re
  File "/usr/lib/python3.5/re.py", line 335, in <module>
    import copyreg
  File "/usr/local/lib/python2.7/dist-packages/copyreg/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
-- PYTHON3_INSTALL_DIR: ${prefix}/
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.5m
-- PYTHON3_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpython3.5m.so
-- CMAKE_SWIG_FLAGS=-c++;-threads;-DSIZE_T_IS_UNSIGNED_INT
CMake Error at python3/CMakeLists.txt:87 (install):
  install TARGETS given no LIBRARY DESTINATION for module target
  "_SoapySDR3".

CMake Error at python3/CMakeLists.txt:92 (install):
  install FILES given no DESTINATION!

-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library , runtime library v0.6.1-g8c5db3e2
 * Tests , library unit tests
 * Docs , doxygen documentation
 * Python , python bindings v2.7.12
 * Python3 , python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 

-- SoapySDR version: v0.6.1-g8c5db3e2
-- ABI/so version: v0.6
-- Install prefix: /usr/local
-- Configuring incomplete, errors occurred!
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeOutput.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package soapysdr:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package soapysdr. Aborting.
root@pi-gsm-01:~/git/SoapySDR/build#

guruofquality, 2017-07-27:
@bookdude It looks like its dying when trying to call into python to get an install prefix. This is basically the underlying command, so it might be useful to figure out why its bombing on your system, I'm curious if it is also erroring out when run manually.

```
 /usr/bin/python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib(plat_specific=True, prefix=''))"
```

anyway, the PYTHON3_INSTALL_DIR is empty because of the error and the configure is broken for that reason. Two work arounds if you cant reproduce and fix the error:

* pass `-DPYTHON3_INSTALL_DIR=desired path` (this should override the automatic one)
* or if not using python3: `-DENABLE_PYTHON3=OFF`

robman501a, 2017-08-10:
This didn't seem to do anything. ' /usr/bin/python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib(plat_specific=True, prefix=''))" '
As for this....
rob@ubuntu:~$ python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages']

I'm not quite sure on where I should put this.
'pass -DPYTHON3_INSTALL_DIR=desired path (this should override the automatic one)' and what path do I use?
or if not using 'python3: -DENABLE_PYTHON3=OFF'

guruofquality, 2017-08-10:
> This didn't seem to do anything. ' /usr/bin/python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib(plat_specific=True, prefix=''))" '

@robman501a Are you able to clone and configure SoapySDR without this error (outside of pybombs, just git clone and mkdir build, cd build && cmake ../) I'm wondering if pybombs overloads the PYTHONPATH environment variable and causes python3 to crash. I don't think python3 is a big use case for pybombs apps, so maybe its ok to disable it in this case. Just a guess, but if cmake fails in pybombs and is ok without it, maybe thats the cause...

> I'm not quite sure on where I should put this.
'pass -DPYTHON3_INSTALL_DIR=desired path (this should override the automatic one)' and what path do I use?
or if not using 'python3: -DENABLE_PYTHON3=OFF'

Its cmake arguments for SoapySDR project, so probably in the relevant pybomb file, I cant advice much more there. but I assume there is a place to add configuration args for cmake projects, since a lot of the pybombs apps are cmake based

robman501a, 2017-08-10:
No issues building SoapySDR outside of PyBombs.

rob@ubuntu:~/SoapySDR/build$ cmake ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") 
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.10") 
-- SWIG_FOUND: TRUE - 3.0.10
-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
-- PYTHONINTERP_FOUND: TRUE - 2.7.13
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.13") 
-- PYTHONLIBS_FOUND: TRUE - 2.7.13
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7
-- PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
-- Found Python3Interp: /usr/bin/python3  
-- Found Python3InterpDbg: /usr/bin/python3-dbg  
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
-- PYTHON3_INSTALL_DIR: ${prefix}/lib/python3/dist-packages
-- Found Python3Libs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so  
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.5m
-- PYTHON3_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library , runtime library v0.6.1-gc772ee1b
 * Tests , library unit tests
 * Docs , doxygen documentation
 * Python , python bindings v2.7.13
 * Python3 , python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 

-- SoapySDR version: v0.6.1-gc772ee1b
[... 270 lines cut ...]
-- Installing: /usr/local/lib/libSoapySDR.so
-- Installing: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/bin/SoapySDRUtil
-- Set runtime path of "/usr/local/bin/SoapySDRUtil" to ""
-- Installing: /usr/local/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/lib/python2.7/dist-packages/_SoapySDR.so
-- Set runtime path of "/usr/local/lib/python2.7/dist-packages/_SoapySDR.so" to ""
-- Installing: /usr/local/lib/python2.7/dist-packages/SoapySDR.py
-- Installing: /usr/local/lib/python3/dist-packages/_SoapySDR.so
-- Set runtime path of "/usr/local/lib/python3/dist-packages/_SoapySDR.so" to ""
-- Installing: /usr/local/lib/python3/dist-packages/SoapySDR.py
rob@ubuntu:~/SoapySDR/build$ sudo ldconfig #needed on debian systems
rob@ubuntu:~/SoapySDR/build$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-gc772ee1b
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6
No modules found!
Loading modules... done
Available factories...null,

guruofquality, 2017-08-10:
That probably means that pybombs is stomping on the PYTHONPATH. Im sure that pybombs is just setting the PYTHONPATH strategically to help it find gnuradio based python libs, and they just do not have to deal with python3 at all. In terms of being a package manager like homebrew, its probably not the best way to go. I'm genuinely curious what the PYTHONPATH is set to, if at all since it seems to be overriding system python packages, not just stuff in /usr/local. Still speculation, none of this is confirmed FWIW.

So anyway, I can think of two things to resolve this:

* Find the maintainer of the soapysdr pybomb and ask them to add `-DENABLE_PYTHON3=OFF` to the cmake arguments. This is probably the easiest and and most straightforward.

* Confirm with pybombs maintainer that it does indeed overwrite the PYTHONPATH, and maybe ask if there is a hook to disable that for certain package configurations or change it to not specify default system install pythons for python

robman501a, 2017-08-10:
Is there a way that I can still use pybombs install gr-gsm and just make it skip the SoapySDR part?

robman501a, 2017-08-10:
'Find the maintainer of the soapysdr pybomb' I have no idea where to look or what to look form.
Would it be a .py file?

guruofquality, 2017-08-10:
I added an issue here: https://github.com/gnuradio/gr-recipes/issues/91

robman501a, 2017-08-11:
Thanks. I wonder how the OP @bookdude is making out with this issue also.

guruofquality, 2017-08-15:
@oe3cts looks like this was your pybombs recipe file, any advice?

guruofquality, 2017-10-05:
Anyway, there is nothing I can do here, its a straightforward build issue. I opened bugs on the correct projects, reported the issue, made a suggestion to either fix the path or to modify the enable flags. Closing. thanks for reporting!

closed 2017-10-05 by guruofquality