pothosware/SoapySDR #133: Building SoapySDR with Python

robman501a, 2017-08-09:
Hi,
I'm trying to make this project but when it comes to SoapySDR, it just keeps on failing. https://github.com/ptrkrysik/gr-gsm/wiki/Installation
I honestly have no clue on what I should do.

rob@ubuntu:~$ sudo pybombs install gr-gsm
[sudo] password for rob:
PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gr-gsm
   |
   \- gr-osmosdr
      |
      +- airspy
      |
      +- hackrf
      |
      \- soapysdr
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: soapysdr
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/soapysdr/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
--
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
-- PYTHONINTERP_FOUND: TRUE - 2.7.13
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.13
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7
-- PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- CMAKE_SWIG_FLAGS=-c++;-threads
--
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
Failed to import the site module
Error processing line 1 of /usr/lib/python3/dist-packages/keyrings.alt-2.2-nspkg.pth:

Traceback (most recent call last):
  File "/usr/lib/python3.5/site.py", line 173, in addpackage
    exec(line)
  File "", line 1, in
  File "/usr/lib/python3.5/types.py", line 166, in
    import functools as _functools
  File "/usr/lib/python3.5/functools.py", line 21, in
    from collections import namedtuple
  File "/usr/lib/python3.5/collections/__init__.py", line 32, in
    from reprlib import recursive_repr as _recursive_repr
  File "/usr/local/lib/python2.7/dist-packages/reprlib/__init__.py", line 7, in
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/site.py", line 580, in
    main()
  File "/usr/lib/python3.5/site.py", line 567, in main
    known_paths = addsitepackages(known_paths)
  File "/usr/lib/python3.5/site.py", line 344, in addsitepackages
    addsitedir(sitedir, known_paths)
  File "/usr/lib/python3.5/site.py", line 212, in addsitedir
    addpackage(sitedir, name, known_paths)
  File "/usr/lib/python3.5/site.py", line 183, in addpackage
    import traceback
  File "/usr/lib/python3.5/traceback.py", line 3, in
    import collections
  File "/usr/lib/python3.5/collections/__init__.py", line 32, in
    from reprlib import recursive_repr as _recursive_repr
  File "/usr/local/lib/python2.7/dist-packages/reprlib/__init__.py", line 7, in
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
-- PYTHON3_INSTALL_DIR: ${prefix}/
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.5m
-- PYTHON3_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- CMAKE_SWIG_FLAGS=-c++;-threads
CMake Error at python3/CMakeLists.txt:87 (install):
  install TARGETS given no LIBRARY DESTINATION for module target
  "_SoapySDR3".

CMake Error at python3/CMakeLists.txt:92 (install):
  install FILES given no DESTINATION!

--
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
--
 * Library , runtime library v0.6.1-gc772ee1b
 * Tests , library unit tests
 * Docs , doxygen documentation
 * Python , python bindings v2.7.13
 * Python3 , python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
--

-- SoapySDR version: v0.6.1-gc772ee1b
-- ABI/so version: v0.6
-- Install prefix: /usr/local
-- Configuring incomplete, errors occurred!
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package soapysdr:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package soapysdr. Aborting.

guruofquality, 2017-08-10:
@robman501a can you look at the reply here: https://github.com/pothosware/SoapySDR/issues/132 I think its the same issue. I dont know why the command is bombing out. I gave some suggestions to investigate it. But worst case, we can always disable python3 if its not being used (not ideal).

Like this shouldnt happen when running import statements in python3, wrong library path, so it could be a system configuration issue:
```
from reprlib import recursive_repr as _recursive_repr
File "/usr/local/lib/python2.7/dist-packages/reprlib/init.py", line 7, in
```

guruofquality, 2017-08-10:
Duplicate: https://github.com/pothosware/SoapySDR/issues/132

closed 2017-08-10 by guruofquality