pothosware/SoapySDR #137: lo_locked sensor always returning "false" for LimeSDR

ccsh23, 2017-09-01:
Hi,

While trying to adapt my app for use od SoapySDR I have noticed that after setting frequency of 1457 MHz for both TX and RX directions on LimeSDR, you cannot confirm that LO have actually locked, altough I have seen on source code of SoapyLMS7 that (at least in theory) reading of corresponding sensor "lo_locked" is actually implemented. I am checking its value in a loop after freq tuning, but its always returns "false" (even after few seconds from calling setFrequency). But when I bypass the loop and insert some sleep command, I am able to confirm that this frequency has been successfully set (via getFrequency call).

Unfortunately today I have no access to devices, so I cannot check if this is happening for any frequency (suppose so) or if it is happening for USRP devices as well. Will report that next week.

Of course, this is low-priority bug, just wanted to let you know about it (or find out what I am doing wrong if that is the case).

guruofquality, 2017-09-03:
Here is a quick one liner to check lock after a tune. FWIW, its printing true for me.

Please give it a try out of curiosity:

`python -c "import SoapySDR; from SoapySDR import *; d = SoapySDR.Device(); d.setFrequency(SOAPY_SDR_RX, 0, 1457e6); print(d.readSensor(SOAPY_SDR_RX, 0, 'lo_locked'))`

ccsh23, 2017-09-04:
Thank you for a quick response, it helped a lot. I don't know why I was not using readSensor() method... Instead, I have been using getSensorInfo() and checking "value" field of returned ArgInfo structure. This way it does not work, but probably should as well?

Anyway, readSensor() works great so I am going to use that. Thank you once again for your help!

guruofquality, 2017-09-04:
I see! good to hear. Think of the value in getSensorInfo() as sort of a default. So if you were populating a GUI widget for example, there would be a sane value to use before actually querying/polling any sensor.

closed 2017-09-04 by guruofquality