pothosware/SoapySDR #138: Timestamps problems: LimeSDR, SoapySDR and UHD...

ccsh23, 2017-09-05:
Dear all,

my project highly relies on precise controlling TX and RX period, i.e. is based on timestamps. So far I have been using USRP N210 (1GbETH-based) and USRP B200 mini (USB3.0-based), but since now I finally got my LimeSDR I would like to rearrange the code so it can also be used. Unfortunately, it seems that both UHD and SoapySDR support of LimeSDR in terms of timestamp-based streaming mode is far from ideal. 

Since I really would like to help as much as I can to fix it, I have developed simple application which is meant to show the issues by transmitting and receiving bursts at the same frequency (like in TDD mode). To make things simpler, it is not meant to benefit from MIMO configuration. I have prepared two versions of this application: one is based on UHD and second is based on SoapySDR API (support for those two is the only thing that was changed in both versions). I have also included configuration files for three different SDR devices: LimeSDR, USRP N210 and USRP B200 mini. This app can be used to show problems occurring when trying to work with timestamps. Here are the links:
https://github.com/ccsh23/UHD_TXRX_Burst_Tester
https://github.com/ccsh23/SoapySDR_TXRX_Burst_Tester

Now, I have tested it and saw following outcomes:
1. UHD version & USRP N210/USRP B200 mini: both RX and TX works just fine. RX calls are returning ERROR_CODE_NONE and TX calls are returning EVENT_CODE_BURST_ACK, as expected. This may be used as a reference.
2. SoapySDR & USRP N210/USRP B200 mini: RX returns STREAM_ERROR for first burst and unexpected TIMEOUTs for all following bursts, TX surprisily seems to work fine, altough NOT_SUPPORTED flag is returned while activating TX stream and while deactivating both TX and RX streams.
3. UHD version & LimeSDR: RX actually seems to work fine, but TX returns TIME_ERROR for first burst (when it should not) and unexpected TIMEOUTs for follwing bursts.
4. SoapySDR version & LimeSDR: RX returns no error but these returns are happening way to often (i.e. if you compare burst timestamp and current timestamp you can see that device did not wait for receive operation as long as it should), TX returns TIME_ERRORs (when it should not, especially for first several bursts) and unexpected OVERFLOWS.

Can you please check this and propose a fix? I am happy to help any way I can. I think that fixing LimeSDR TX problem in UHD seems like the easiest way to go (see 3.), because in case of SoapySDR there are different problems with all devices, which will probably take more time to solve.

I have not seen any truly working timestamps-based TX&RX example for LimeSDR, and mentioned problems are probably the cause. Hopefully we will be able to fix it :)

guruofquality, 2017-09-05:
> I have not seen any truly working timestamps-based TX&RX example for LimeSDR, and mentioned problems are probably the cause. Hopefully we will be able to fix it :)

There just might be some differences and I hope to work my way through them. To reply quickly, if it helps, can you try the following python example with limesdr?

https://github.com/pothosware/SoapySDR/blob/master/python/apps/MeasureDelay.py

`python MeasureDelay.py --rate=10e6 --freq=1e9 --rxGain=20 --txGain=20
--rxAnt=LNAL --txAnt=BAND1`

It uses a timed transmit burst and timed reception to measure a pulse delay. In the above example I just put some rubber duck antennas on channel A LNAL and BAND1 ports. Also make sure to use the latest stable/master branch as there were some recent fixes.

guruofquality, 2017-09-06:
@ccsh23 Alright, here is the breakdown. Also I encourage you to look at [activate/deactivate/read/writeStream in SoapyUHD](
https://github.com/pothosware/SoapyUHD/blob/master/SoapyUHDDevice.cpp#L240). Some of these issue may have a good explanation, others we might change because its just better convention. Here is the breakdown, let me know what you think:

> RX returns STREAM_ERROR for first burst and unexpected TIMEOUTs for all following bursts

This may be a late command, which would also explain the timeouts. 

```
case uhd::rx_metadata_t::ERROR_CODE_LATE_COMMAND: return SOAPY_SDR_STREAM_ERROR;
case uhd::rx_metadata_t::ERROR_CODE_BROKEN_CHAIN: return SOAPY_SDR_STREAM_ERROR;
```

> TX surprisily seems to work fine, altough NOT_SUPPORTED flag is returned while activating TX stream and while deactivating both TX and RX streams.

For tx, there is no equivalent deactivateStream in uhd, and we intentionally return NOT_SUPPORTED. That said, returning 0 as a successful "NOP" call might be preferable here. Its not really an error to call deactivateStream(), we want to encourage it for portable code because some implementations may use it.

Rx deactivateStream() can return NOT_SUPPORTED if there are any flags set. I think this is wrong because it could contain a time flag and timestamp to pass to issue_stream_cmd.

```
if (flags != 0) return SOAPY_SDR_NOT_SUPPORTED;

//stop the stream
stream->rx->issue_stream_cmd(uhd::stream_cmd_t::STREAM_MODE_STOP_CONTINUOUS);
```

> UHD version & LimeSDR: RX actually seems to work fine, but TX returns TIME_ERROR for first burst (when it should not) and unexpected TIMEOUTs for follwing bursts.

For LimeSDR there was a recent fix for flushing out small bursts, I would update LimeSuite and try again: https://github.com/myriadrf/LimeSuite/commit/46764a32a0c3ef698c9e5098025017f249ee6101

For UHD, writeStream is almost a perfect translation to stream->send(). So I dont know what could be wrong, but from the other issue, it sounded like you might have a bug in your examples with timestamps.

> SoapySDR version & LimeSDR: RX returns no error but these returns are happening way to often (i.e. if you compare burst timestamp and current timestamp you can see that device did not wait for receive operation as long as it should),

I can check LimeSuite. But I still suspect timestamp issues in the example. I can take a look at that too later too

> TX returns TIME_ERRORs (when it should not, especially for first several bursts) and unexpected OVERFLOWS.

Between the recent fix in limesuite and I suspect a timestamp issue in your examples, I think we can fix the time errors being reported.

But bizarrely the overflows are actually reported through the stream status in limesuite. You can see it [here as packet dropped](https://github.com/myriadrf/LimeSuite/blob/master/src/protocols/ILimeSDRStreaming.cpp#L83). I think that the overflows should be reported as a flag that is synchronous with the rx stream. So I think this can be fixed but we would have to make sure that other apps arent impacted. You might have to ignore this return code in the meantime.

ccsh23, 2017-09-06:
Dear Josh,

thank you for quick response. I got ca. 8 us of burst delay with your python app.

I have builded LimeSuite from source again with the newest fix, but unfortunately it does not solve the problem for TX bursts support with LimeSDR. To be honest I doubt that there is some „timestamp issue” in my UHD-based example. I have some experience  with UHD, as I have been sucessfully using it to implement TDD OFDM-based radio link, obviously based on timestamps. SoapySDR API is new thing for me, so there migh be some bugs in SoapySDR-based example, but I really believe that UHD version is correct. After all, it works like a charm with USRP B200 mini and USRP N210. So I still suppose that something is rather wrong with TX timestamps handling by LimeSDR. For some reason it reports TIME_ERROR for the very first burst when I try to transmit burst with timestamp which is even several seconds in future. This is obiviously not to soon to use send() with such timestamp, so what is happening here?

> activateStream() is the mechanism to requests bursts. The timestamp is only an output parameter for readStream()

As for SoapySDR-based example, I have to admit that I feel lost now. Lets assume that I want to send 100 samples (small number so it will fit in a single buffer) every 100 miliseconds or receive 100 samples every 100 miliseconds. Which functions and flags should I use and where to put timestamps? Can you confirm that this is the right approach or give some hint which would be the right one?

TX:
```
tx_stream = device->setupStream(SOAPY_SDR_TX, "CF32", ...);
device->activateStream(tx_stream); //NO FLAGS OR TIMESTAMPS HERE!
long long timestamp = device->getHardwareTime() + 1e9; //1 SECOND IN FUTURE
while (true)
{
	device->writeStream(tx_stream, buffs,  nsamps, SOAPY_SDR_HAS_TIME | SOAPY_SDR_END_BURST, timestamp); //ACTUAL TX FLAGS AND TIMESTAMP HERE!
	timestamp += 0.1e9; //+100 ms
}
device->deactivateStream(tx_stream); //NO FLAGS OR TIMESTAMPS HERE!
device->closeStream();

```
RX:
```
device->setupStream(SOAPY_SDR_RX, "CF32", ...);
long long timestamp = device->getHardwareTime() + 1e9; //1 SECOND IN FUTURE
while (true)
{
	device->activateStream(rx_stream,  SOAPY_SDR_HAS_TIME | SOAPY_SDR_END_BURST, timestamp, nsamps); //ACTUAL RX FLAGS AND TIMESTAMP HERE!
	device->readStream(rx_stream, buffs, nsamps, read_flags, read_timestamp); //THESE FLAGS AND TIMESTAMP ARE GOING TO BE SET BY DEVICE
	device->deactivateStream(rx_stream); //NO FLAGS OR TIMESTAMPS HERE!
	timestamp += 0.1e9; //+100 ms
}
device->closeStream(rx_stream);

```
If this is right, then I am afraid it still does not work how it should with LimeSDR (TX: TIME_ERRORS and OVERFLOWS, RX: receiving smaller number of samples than requested), works okay with USRP B200 mini, but in case of USRP N210 some rx calls return 0 number of samples with no error.

Thank you once again for help, I really appreciate that!

guruofquality, 2017-09-06:
* For Tx this looks basically correct. :-)

* For Rx, dont call deactivateStream(), this is for stopping continuous streaming, but when using bursts, the stream is automatically stopped. deactivateStream() could actually cause an additional packet in some cases.

* Also for Rx, since the time delta is large, about a second, the call could timeout. Make sure to pass a larger timeout in if this is the case.

* And as usual, check the return codes. For small buffers, read/write can consume the entire buffer, but larger ones may need multiple calls and incrementing the buffer pointer. Just like a unix sock read/write().

> but in case of USRP N210 some rx calls return 0 number of samples with no error.

I think in this case the underlying stream->recv() from UHD is retuning 0. Thats a little weird, but it might be the result of calling deactivateStream(). Its possible that it triggers a mini end of burst packet.

https://github.com/pothosware/SoapyUHD/blob/master/SoapyUHDDevice.cpp#L277

Just to clear up any confusion about some of these calls, if you are familiar with UHD, they are basically just wrappers for each other, although with different timestamp units:

* readStream calls stream->recv()
* writeStream calls stream->send()
* activateStream calls stream->issue_stream_command() for rx streams
* deactivateStream calls stream->issue_stream_command(STREAM_MODE_STOP_CONTINUOUS) for rx streams
* and de/activateStream() is a NOP for tx streams in uhd

The calls above should basically work the same on LimeSDR. And it looks like the MeasureDelay.py was working for you. This example is essentially doing the same thing, demoing a small burst with timestamps in both rx and tx directions. So I hope it helps as a reference point. -- Since it sounds like SoapyUHD is basically working for you, perhaps I can run the same example on LimeSDR later tonight and see what happens.

ccsh23, 2017-09-07:
Thanks again for quick answer. I have updated the code according to your suggestions (my github contains already updated version) but still no luck with LimeSDR. I would appreciate if you could find some time to see what is happening here, because being honest I am running out of ideas. Meantime, I got some further questions:

> For Rx, dont call deactivateStream(), this is for stopping continuous streaming, but when using bursts, the stream is automatically stopped. deactivateStream() could actually cause an additional packet in some cases.

I have removed it from RX loop as you have suggested, but still need to put it in the end (in desctructor of my class, right before closeStream), otherwise I got "[ERROR] transfer no device" and core dump. I am a little confused since you said that is should not be called for burst-based streaming. So is that suppose to happen?

> 
And as usual, check the return codes. For small buffers, read/write can consume the entire buffer, but larger ones may need multiple calls and incrementing the buffer pointer. Just like a unix sock read/write().

I think this is actually significant difference comparing to UHD, where single call to send() or recv() is enough even if RX or TX stream MTU is smaller than requested number of samples to be transmitted or received (for example, in case of USRP B200 mini TX&RX MTU is 2044 samples, but I am able to send or receive much larger number of samples, say 9999, per single send() or recv() call). Not that this is problem of any kind, but just wanted to know what should I expect when using UHD calls with LimeSDR. So far, to avoid problems coming from that difference, I have reduced burst length so that I am trying to transmit or receive only 100 samples.

> > but in case of USRP N210 some rx calls return 0 number of samples with no error.
> 
> I think in this case the underlying stream->recv() from UHD is retuning 0. Thats a little weird, but it might be the result of calling deactivateStream(). Its possible that it triggers a mini end of burst packet.

As I mentioned, I have removed deactivateStream calls from RX loop. Going to test if it helped on Monday, because so far I got no access to N210 device.

Now actually the essential question for today is as follows: at which point hardware timer starts on LimeSDR? I am asking because I have made simple test to see how it increments after all settings (like antennas, gains, frequencies etc.) are set, both TX and RX streams are set up and TX stream has been activated. At this point I put following code (first version for UHD, second for SoapySDR):
```

device->set_time_now(uhd::time_spec_t(0.0f),0);

for (int i=1; i<=5; i++)
{
	sleep(1);
	long long hardware_time = 1e9 * device->get_time_now().get_real_secs();
	cout << endl << "hardware time after " << i << " seconds from reset: " << hardware_time;
}
```
```
device->setHardwareTime(0);

for (int i=1; i<=5; i++)
{
	sleep(1);
	long long hardware_time = device->getHardwareTime();
	cout << endl << "hardware time after " << i << " seconds from reset: " << hardware_time;
}
```
Now, for USRP B200 mini (both in UHD and SoapySDR-based versions) I got expected output:
hardware time after 1 seconds from reset: 1000146558
hardware time after 2 seconds from reset: 2000355336
hardware time after 3 seconds from reset: 3000621744
hardware time after 4 seconds from reset: 4000884072
hardware time after 5 seconds from reset: 5001144720

However for LimeSDR (both in UHD and SoapySDR-based versions) it looks like it does not run at all:
hardware time after 1 seconds from reset: 0
hardware time after 2 seconds from reset: 0
hardware time after 3 seconds from reset: 0
hardware time after 4 seconds from reset: 0
hardware time after 5 seconds from reset: 0

It that correct? From what I can see, it starts to increment only when RX calls are happening. If so, debugging TX-only apps might be very difficult since you got no way to check what is the current timestamp on a device, thus you can not check if your command is late, for example.

As always, your help would be greatly appreciated.

guruofquality, 2017-09-07:
> I have removed it from RX loop as you have suggested, but still need to put it in the end (in desctructor of my class, right before closeStream), otherwise I got "[ERROR] transfer no device" and core dump. I am a little confused since you said that it should not be called for burst-based streaming. So is that suppose to happen?

I see. Well you definitely don't want to have deactivateStream() at the end of every burst. In this case calling it once before closing the stream is a good idea. It turns out that on LimeSDR de/activateStream() control some threads -- I think that the end of the day closeStream() on LimeSDR should actually make sure all threads are stopped regardless. I will try to make that modification.

> I think this is actually significant difference comparing to UHD, where single call to send() or recv() is enough even if RX or TX stream MTU is smaller than requested number of samples to be transmitted or received (for example, in case of USRP B200 mini TX&RX MTU is 2044 samples, but I am able to send or receive much larger number of samples, say 9999, per single send() or recv() call). Not that this is problem of any kind, just wanted to know what should I expect when using UHD calls with LimeSDR. So far, to avoid problems coming from that difference, I have reduced burst length so that I am trying to transmit or receive only 100 samples.

Broadly speaking any of these stream APIs is free to consume or produce as much as the buffer as it wants unless the special SOAPY_SDR_ONE_PACKET (or uhd equivalent) flag is passed. And it depends not only on the device MTU, but the timeout being used, and underlying implementation. I'm surprised to hear about the UHD sizes being small (2044 samples) because I thought there was an internal loop to copy the samples out over multiple buffers. In any case, LimeSDR has an internal fifo which is quite large and its simply copying as much as the buffer as it can, provided there is space to perform the copy.

> However for LimeSDR (both in UHD and SoapySDR-based versions) it looks like it does not run at all:

Sorry, its a limesdr caveat. The rxStream must be setup in order to get timestamps. The timestamps are not read from a register, but rather synced over every rx packet. So you dont notice this in a Tx/Rx app, but in an Tx only app the rx stream must be setup for this reason.

ccsh23, 2017-09-07:
> Sorry, its a limesdr caveat. The rxStream must be setup in order to get timestamps. The timestamps are not read from a register, but rather synced over every rx packet. So you dont notice this in a Tx/Rx app, but in an Tx only app the rx stream must be setup for this reason.

Okay, but does it mean that I can use only these TX timestamps which were passed with RX calls as well? Say that my RX bursts are beginning at timestamps +1, +3, +5, +7 seconds and I want to transmit TX bursts at timestamps +2, +4, +6, +8 seconds (i.e. different than RX timestamps). Should that be possible on LimeSDR? I am asking because I don't know in details how this synchronization of timestamps to RX packets works. If this should work, then I really have no idea what is the reason of my problems...

guruofquality, 2017-09-07:
> You probably meant activated?

Yes, that starts the receiver thread. Sorry my mistake

>  Anyway, does it mean that I can use only these TX timestamps which were passed with RX calls as well? Say that my RX bursts are beginning at timestamps +1, +3, +5, +7 seconds and I want to transmit TX bursts at timestamps +2, +4, +6, +8 seconds (i.e. different than RX timestamps). Should that be possible on LimeSDR? I am asking because I don't know in details how this synchronization of timestamps to RX packets works. If this should work, then I really have no idea what is the reason of my problems...

Yea, you should be able to do that. The times are on the exact same counter inside the LimeSDR. Receive at second 1 and transmit at second 2 is fine.

What I meant to say is that the implementation of getHardwareTime(), it doesnt read a time register in the FPGA like you might expect, instead, its just a cache of the most recent time seen by the internal receiver thread. The value is just as good a if there was a register read, but if the thread is not spinning, then the software does not know approximately the hardware time.

hardware time is generally used to get an approximation of the time to send something in the future or request a burst in the future. But if you are scheduling an transmit frame based on a receive time from readStream(), then getHardwareTime() is not even relevant.

ccsh23, 2017-09-08:
Okay, got it. It's good information that it should be possibile, little worse that for some reason I cannot got it work yet :P But I think I might got some trail: in my code first TX burst is happening before first RX burst. Maybe that messes something up (altough following TX calls are happening after first RX call, so even if first TX burst would result in error, the following should probably be handled correctly). Still, I am gonna make sure that RX will be called first and let you know about the results.

Thanks again for your time!

ccsh23, 2017-09-08:
I have changed the code to make sure that TX loop will fire after RX one, I am also sure that first TX burst timestamp is later than first RX burst timestamp - still does not help.

Since even with 100 samples long buffers I got problems even with just receiving them entirely in SoapySDR version, so far I think UHD is more stable (at least receiving operations work correctly here). Unfortunately, TX is still the problem and I really believe that device reactions are wrong.

Take a look at debug information about first TX burst I am trying to transmit:

`[TX]  current_time=360284404, burst_time: 1200000003, ack_code: EVENT_CODE_TIME_ERROR, no_of_transmitted_samples: 100/100`

As you said, "current_time" is just an estimation, but since RX thread is providing new data with every 100ms without any errors, we can assume that this "current_time" is correct to that extent. Which means that current time is about +360ms claimed (+/- 100ms because of caching) and I am requesting TX burst to be transmitted at +1.2s. This is obiviously a lot of time for device to work with, yet it returns TIME_ERROR in ack, suggesting that this command is late (well, it is not!).

Another example of incorrect behavior (this is actually second TX burst I am trying to transmit):

`[TX]  current_time=361263604, burst_time: 1300000000, ack_code: ERROR_CODE_TIMEOUT, no_of_transmitted_samples: 100/100`

Again, device current time is about +361ms claimed (+/- 100ms because of caching), I am trying to schedule a burst at +1.3s and it returns TIMEOUT. Since I gave timeout parameter of 2 seconds for both send() and recv_async_msg() calls, it is again incorrect.

Can you please take a detailed look what is happening here?

guruofquality, 2017-09-09:
@ccsh23 Did you catch my comments here: https://github.com/ccsh23/SoapySDR_TXRX_Burst_Tester/issues/1 -- Theres no EOB ack indication on the stream status. My concern is that the readStreamStatus is waiting for maximum timeout of 2 seconds because its waiting on a message that will never come.

Secondly, there seems to be a spurious late packet reported when the rx stream is started up. I will ask the Lime guys about that. Its probably some start up condition in the FPGA core. And I think it might be throwing you off.

But basically, I made the following modification to your app, a smaller stream status timeout and reporting ok for timeout status and it seems to be transmitting without any late error messages. and for that matter not "L"s printed by the tx thread.

```
@@ -422,7 +422,7 @@ namespace sdr
                                        chan_mask,
                                        flags,
                                        burst_time,
-                                       1e6 * device_cfg->T_timeout
+                                       1000
                                );
 
                                switch(stream_status)
@@ -458,7 +458,7 @@ namespace sdr
 
                                tx_verbose_msg += (boost::format(", no_of_transmitted_samples: %u/%u") % no_of_transmitted_samples % no_of_requested_samples).str();
 
-                               if ((stream_status == 0) && (no_of_transmitted_samples == no_of_requested_samples))
+                               if ((stream_status == SOAPY_SDR_TIMEOUT) && (no_of_transmitted_samples == no_of_requested_samples))
                                {
                                        msg("sdr: " + tx_verbose_msg);
                                        res = true;
```

Also I am curious what you are actually seeing over the air. Other than the differences with the stream status, the measure delay example is actually working over the air for you, and so I think that your application may be working too.

I don't know how else to help, but to summarize:

* There is no burst ack for stream status. If you need to wait for that event, the rx timestamp might be an alternative option. I think this would be a useful feature, but from what I recall the developer did not want to add it. and it would require FPGA changes.

* There is an initial spurious late indication. I can see it in an rx only application. So its probably a bug to be reported and fixed. My one suggestion is to either ignore it the first time, or to readStreamStatus once the rxStream is activated the first time just to flush it out.

* I'm happy to support SoapySDR projects in general or what should be good API semantics on this issue tracker, but I think anything hardware specific should really go to that hardware's tracker. That way other developers on that project might also be able to offer some advice. https://github.com/myriadrf/LimeSuite/issues would be ideal for anything further

ccsh23, 2017-09-11:
Dear Josh,

Thank you again for quick response. As always I really appreciate your help. 

I also agree that this is not the best place for hardware-specific problems and would be happy to move it somewhere else. Assuming that I am going to continue working in UHD with LimeSDR, I guess the best place to open new issues (such as e.g. wrong value of returned gain) would be SoapyUHD github page, right? Or maybe LimeSuite?

But as for your last response, I would just like to be sure that I got it right:

> There is no burst ack for stream status. If you need to wait for that event, the rx timestamp might be an alternative option. I think this would be a useful feature, but from what I recall the developer did not want to add it. and it would require FPGA changes.

Does it apply only to async acks (so you cannot make sure that your burst was indeed transmitted entirely with no error) or there is no way at all to make also send() command block until device timestamp will match the one requested by the user? What I mean is that for now it seems like send() is returning immediately always with no error, even for very large burst period (providing tx timestamps which are greater from the previous by 1 seconds for example - I was expecting it to be blocking here) and it also does not return any error if evident late command is happening (I have intentionally been giving far past timestamps in some iterations, and it was still returning the same number of samples which were provided, indicating that everything is fine). So it seems like it is impossible to create simple loop with send() command inside which will be polled by rate determined by device clock. To be honest, for now it seems like absolutely no way to be sure that device is transmitting anything at all (send() is working like "ok, I will try to do it but I don't promise anything and don't care if your timestamp is already from 1990s"). So is that what you meant by rx timestamp as only alternative option? In that sense that I should put both recv() and send() commands inside the same loop so it will be polled by blocking recv() command, thus making also non-blocking send() command to be executed at the same rate, determined by device clock?

Again, sorry for creating this issue here, I promise I will ask further questions wherever you will point :)

guruofquality, 2017-09-13:
> I also agree that this is not the best place for hardware-specific problems and would be happy to move it somewhere else. Assuming that I am going to continue working in UHD with LimeSDR, I guess [...]

Yea I think that works. Certainly LimeSuite github issue tracker if its related to the limesdr support, or SoapyUHD issue tracker if perhaps there is a specific problem there.

> Does it apply only to async acks (so you cannot make sure that your burst was indeed transmitted entirely with no error) or there is no way at all to make also send() command block until device [...]

Most of these devices have large transmit fifos (usrp and limesdr included):

* in the PC (driver/kernel)
* in the fpga/fx3/etc

Sending samples basically writes into an underlying fifo, the call returns asap, and the underlying hardware and driver handles sample delivery. Fundamentally this is how network sockets work too. The only time these write/send calls "block" is when underlying fifo fills up or flow control throttles.

* If there is a timestamp, it causes the FPGA to block the stream until its internal counter matches.
* If there is no timestamp, the internal FPGA cores samples as soon as they are available

In regards to how can we know that the samples were transmitted?

* Some USRP devices support a async burst acknowledgement. This requires hardware support, some find it useful, in some cases if you wait for this ack, you may have wasted time where other processing could have happened. It all depends
* LimeSDR or BladeRF for example does not support async burst acknowledgement.

There is another way to know when a transmit burst completed that works a little more universally:

* A new timestamp is always coming in through readStream(). So when rxTimestamp > txBurstTime, you know the burst occurred. Most trx apps seem to either use rx timestamps or rx events (like a frame correlation) to schedule tx events.
* Tx is almost always a massive fifo, so you have to manage when to feed him samples. You need a concept of device time. This can come from rx in most cases, but it could come from getHardwareTime() too

> Again, sorry for creating this issue here, I promise I will ask further questions wherever you will point :)

No worries. Happens all the time. Just trying to keep things organized from my end, and make sure the right people know about a potential problem.

guruofquality, 2017-09-29:
Looks like we resolved some misconceptions, fixed a few bugs, and moved remaining discussion to myriadrf. So thanks!

closed 2017-09-29 by guruofquality