pothosware/SoapySDR #141: airspy hf+

dgorissen, 2017-09-29:
Any concrete plans to support the airspy hf+ or would this fold under the existing airspy driver?

guruofquality, 2017-09-29:
> would this fold under the existing airspy driver

I think folding it into the existing [SoapyAirspy](https://github.com/pothosware/SoapyAirspy) depends on if the existing airspy host driver supports both devices or if there is a new library altogether.

> Any concrete plans to support the airspy hf+

If its not part of libairspy, then we are looking for volunteers with both hardware and a compiler :-)

dgorissen, 2017-09-29:
Cool. Once preorders are announced I will book one and get back in touch once its in my hands ;)

nmaster2042, 2017-11-12:
The airspy HF+ isn't supported by the existing airspy driver

Here is the driver for airspy HF+: https://github.com/airspy/airspyhf

Pre orders are opened now

stottsc, 2018-01-13:
I have created a SoapySDR plugin for the Airspy HF+ based on the plugin for the original Airspy. I have done only basic testing so far, on Ubuntu 17.10, with CubicSDR. I've only tested on the HF input, with an HF receive antenna.

[https://github.com/stottsc/SoapyAirspyHF](https://github.com/stottsc/SoapyAirspyHF)

guruofquality, 2018-01-14:
@stottsc very awesome. Its always great to see the community contributing new hardware support!

I will extend the usual offer, If you would like the project hosted here I will be happy to help with the move and the account/organization R/W access to the repo. Otherwise, its just as good where it is, so be it. -- Either way, let me know. I would like to update url links in the sidebar, wiki, things like that. And add this to the packaging efforts if possible.

stottsc, 2018-01-17:
@guruofquality Would hosting it with the other SoapySDR plugins make it easier to incorporate into packaging, etc? If so, maybe we should move it here. It would be nice to make the Airspy HF+ support as easy for people to get as the support for other hardware.

guruofquality, 2018-01-18:
Yea, sure I think it makes it easier on a number of levels. Hosting it here is my preference (actually hosting it with the manufacturer's github or even in the airspy driver github would be pretty cool too). Anyway, if you are down with that, just move the repo to @guruofquality in the repository settings, and I can move it here pothosware/ and setup access permissions. -- Oddly enough you cant move repos to organizations directly (last time I tried) unless you have admin access, so we have to do this extra hop between accounts.

streamofstars, 2018-01-24:
@stottsc any instructions on compiling https://github.com/stottsc/SoapyAirspyHF please? I have SoapySDR installed already, https://github.com/airspy/airspyhf as well.

stottsc, 2018-01-24:
mkdir build
cd build
cmake ..
make
sudo make install

<https://github.com/pothosware/SoapyAirspy/wiki#probing-soapy-airspy>
<https://github.com/pothosware/SoapyAirspy/wiki#id4>

streamofstars, 2018-01-24:
@stottsc thanks. What is worth mentioning for people like me who are not software developers or even do not compile others' stuff too often, one needs to have SoapySDR built from sources (see https://github.com/pothosware/SoapySDR/wiki/BuildGuide). I had it previously installed from binaries and was not able to build https://github.com/stottsc/SoapyAirspyHF

Anyway, it does not seem to work for me. Before I started with SoapyAirspyHF in dmesg I could see it detects the device
```
[ 1181.654312] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[ 1181.787154] usb 3-2: New USB device found, idVendor=03eb, idProduct=800c
[ 1181.787171] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1181.787180] usb 3-2: Product: AIRSPY HF+
[ 1181.787188] usb 3-2: Manufacturer: www.airspy.com
[ 1181.787195] usb 3-2: SerialNumber: AIRSPYHF SN:<serial-no-redacted>
```
now it is just
`[  589.555369] usb usb3-port2: Cannot enable. Maybe the USB cable is bad?`

stottsc, 2018-01-24:
I'm not a software developer either, just an interested hobbyist. Not sure
to what your error message is related. That looks like an error from the
USB stack, not libairspyhf or SoapySDR/SoapyAirspyHF. Have you successfully
tested your hardware setup with any other software? Your hardware or USB
cable may have an issue.

guruofquality, 2018-02-07:
The repo is now here: https://github.com/pothosware/SoapyAirspyHF/wiki Let me know if there is some issue. I will try to get it into some of the packaging efforts in the coming weeks. Thanks @stottsc  !

Lets do any further issues on the SoapyAirspyHF tracker

closed 2018-02-07 by guruofquality