pothosware/SoapySDR #143: ImportError: No module named 'SoapySDR'

FFY00, 2017-10-20:
My build proccess.

```
anubis@FFY00:~/git/SoapySDR/build$ cmake .. -DUSE_PYTHON_CONFIG=ON
-- The C compiler identification is GNU 5.4.1
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.11") 
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.8") 
-- SWIG_FOUND: TRUE - 3.0.8
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- PYTHONINTERP_FOUND: TRUE - 2.7.12
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.12
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7;/usr/include/x86_64-linux-gnu/python2.7
-- PYTHON_LIBRARIES: -L/usr/lib/python2.7/config-x86_64-linux-gnu -L/usr/lib -lpython2.7 -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.8
-- Found Python3Interp: /usr/bin/python3  
-- Could NOT find Python3InterpDbg (missing:  PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
-- PYTHON3_INSTALL_DIR: ${prefix}/lib/python3/dist-packages
-- Found Python3Libs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so  
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.5m
-- PYTHON3_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library , runtime library v0.6.1-gba9ad2d2
 * Tests , library unit tests
 * Docs , doxygen documentation
 * Python , python bindings v2.7.12
 * Python3 , python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 

-- SoapySDR version: v0.6.1-gba9ad2d2
[... 291 lines cut ...]
  File "/home/anubis/.local/lib/python3.5/site-packages/soapypower/__main__.py", line 5, in <module>
    import simplesoapy
  File "/home/anubis/.local/lib/python3.5/site-packages/simplesoapy.py", line 5, in <module>
    import SoapySDR
ImportError: No module named 'SoapySDR'
```

If I run it as a privileged user, I get.

```
Traceback (most recent call last):
  File "/usr/local/bin/soapy_power", line 9, in <module>
    load_entry_point('soapy-power==1.6.1', 'console_scripts', 'soapy_power')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "build/bdist.linux-x86_64/egg/soapypower/__main__.py", line 5, in <module>
  File "build/bdist.linux-x86_64/egg/simplesoapy.py", line 3, in <module>
ImportError: No module named abc
```

guruofquality, 2017-10-20:
set the PYTHONPATH environment variable

FFY00, 2017-10-20:
I already did that. I tried with both python 2 and 3 paths.

guruofquality, 2017-10-20:
Do this: `python3 -c "import sys; print(sys.path)"` and see if the path contains /usr/local/lib/python3/dist-packages

The install path can also be overridden with -DPYTHON3_INSTALL_DIR=foo to cmake

FFY00, 2017-10-20:
It does.

FFY00, 2017-10-20:
Thanks, it's working.

closed 2017-10-20 by FFY00

Dantali0n, 2019-01-24:
Would there be a way to better detect the sys path of python and automatically adjusting the PYTHON3_INSTALL_DIR as it seems that /usr/lib/pythonxx is the default on Arch based systems while the Soapy is installed into /usr/local/lib/pythonxx. 

This was at least my experience when using Manjaro 18.

guruofquality, 2019-01-24:
> Would there be a way to better detect the sys path of python and automatically adjust the PYTHON3_INSTALL_DIR as it seems that /usr/lib/pythonxx is the default on Arch based systems while the Soapy is installed into /usr/local/lib/pythonxx.

I would check what site.getsitepackages() is showing. If there is no option to use /usr/local, thats unfortunate, but we still have the option of setting CMAKE_INSTALL_PREFIX, or setting PYTHONPATH, or an entry in an installed .pth file

MartGro, 2023-02-12:
I was encountering the same error when trying to import SoapySDR in a conda environment. 
I solved it by adding the path to the library to the conda environment.
1)  `conda activate my_env`
2)` conda develop /usr/local/lib/python3.8/site-packages`  <-- path to the library. This is shown when running cmake

Then it works for me. I had to install conda-build beforehand (`conda install conda-build`)