pothosware/SoapySDR #144: SoapySDR runtime installer for Windows

philburr, 2017-10-20:
I'm looking at porting my SDR software to windows.  It currently uses SoapySDR exclusively as the abstraction layer around hardware.  However, I'm a bit dismayed that there is no separate SoapySDR runtime installer for windows.  In order to get the SoapySDR runtime on windows, my users would have to install the entire pothos software bundle.

What would be the possibility of getting an installer just for the SoapySDR runtime and drivers?  My project is a 500k download.  The Pothos installer is almost 80Meg.

guruofquality, 2017-10-20:
Is the installer download size the issue or just the installation size? Because there are component specific checkboxes for the installer, so you can disable a lot of stuff. And in that vein, we could also add some more component options for just SDR driver stuff.

philburr, 2017-10-20:
No I think it is more of a hassle than anything else.  What would be ideal would be a distinct installer that I could embed in and run from my own installer.  Asking a user to wade through dialogs, make certain to uncheck everything except this or that is a pain.

For a vendor neutral product, what would be ideal is just an installer that installed SoapySDR as a standalone runtime and drivers.  I'm thinking along the lines of the MSVC runtime.  Just install, nothing else.

guruofquality, 2017-10-20:
Makes sense. https://github.com/pothosware/PothosSDR/issues/42