pothosware/SoapySDR #145: Can't figure out    SoapySDRDevice_readStream()

navysurfer, 2017-11-10:
I'm having issues with the setup.  Right now I'm just trying to debug it to get it working.  I have the program read from the device, and send the buffer to stdio.  I then pipe it to another program to decimate it down and demodulate it from there, eg: `bin/Release/SDR | csdr convert_s16_f | csdr fmdemod_quadri_novect_cf | aplay -r 48000 -f FLOAT_LE`

I know the other program works as expected because I can pipe rtl_sdr to it without issue.
Can you please take a look at my code and help me figure out what I've done wrong, or what I'm missing.  I've tried to make it as basic as possible.

Header:
```
#ifndef RADIO_H
#define RADIO_H

//#include <SoapySDR/Config.hpp>
//#include <SoapySDR/Types.hpp>
//#include <SoapySDR/Constants.h>
//#include <SoapySDR/Version.hpp>
//#include <SoapySDR/Modules.hpp>
//#include <SoapySDR/Registry.hpp>
//#include <SoapySDR/Errors.hpp>

#include <limits.h>

///#include <cstdlib>
///#include <cstddef>
#include <iostream>
///#include <getopt.h>
///#include <vector>
///#include <chrono>
#include <unistd.h>
#include <string.h>
///#include <bitset>
///#include <libhackrf/hackrf.h>

extern "C" {
    #include <SoapySDR/Device.h>
    #include <SoapySDR/Formats.h>
}

//#define DEFAULT_BUF_LENGTH		(16 * 16384)
#define DEFAULT_BUF_LENGTH		(2048)

class Radio
{
    public:

        Radio();
        int initialize_radio();
        int getStream(float **output, int &outSize);
        void uninit_radio();
        size_t elemSize;
        size_t numElems;
        //std::vector<void *> buffs;
        int16_t *buffer;
        uint8_t *buf8 = NULL;
        SoapySDRDevice *device;
        SoapySDRStream *stream;
        ///std::chrono::high_resolution_clock::time_point startTime;
        ///std::chrono::high_resolution_clock::time_point timeLastPrint;
        ///std::chrono::high_resolution_clock::time_point timeLastSpin;
        ///std::chrono::high_resolution_clock::time_point timeLastStatus;

        //hackrf_device *hackrf;

    protected:

    private:
        FILE *file;
        char *output_format = SOAPY_SDR_CU8;
        /*struct fdata {
            std::complex<float> imag;
            float real;
        };*/
        uint32_t out_block_size = DEFAULT_BUF_LENGTH;
        int verbose_device_search(char *s, SoapySDRDevice **devOut, SoapySDRStream **streamOut, const char *format);
        int suppress_stdout_start(void);
        void suppress_stdout_stop(int tmp_stdout);
        int flags = 0;
        long long timeNs = 0;
        int n_read = 0;
        float *outbuff;
        long timeoutNs = 10000000;
};

#endif // RADIO_H
```

Source:
```
#include "Radio.h"

Radio::Radio()
{
    //ctor
}

int Radio::initialize_radio(){
    file = stdout;
    int tmp_stdout = suppress_stdout_start();
	buffer = (int16_t*)malloc(out_block_size * SoapySDR_formatToSize(SOAPY_SDR_CS16));
    //buf8 = (uint8_t*)malloc(out_block_size * SoapySDR_formatToSize(SOAPY_SDR_CS8));

    buf8 = (uint8_t*)malloc(out_block_size * SoapySDR_formatToSize(SOAPY_SDR_CU8));
        int r = verbose_device_search("", &device, &stream, SOAPY_SDR_CS16);
        if(r != 0){
            printf("Couldn't find SDR Device\n");
            exit(0);
        }
        //SoapySDRDevice_writeSetting(device, "direct_samp", )

        //device = SoapySDR::Device::make("driver=hackrf");
        //device = SoapySDR::Device::make("rtlsdr");
        //SoapySDRKwargs args = {};
        //SoapySDRKwargs_set(&args, "driver", "rtlsdr");
        //device = SoapySDR::Device::make(args);

        int direction = -1;
        double sampleRate = 2400000.0;
        fprintf(stderr, "Reading at Sample Rate: %d\n", (long)sampleRate);
        r = SoapySDRDevice_setSampleRate(device, SOAPY_SDR_RX, 0, (double)sampleRate);
        if(r != 0){
            printf("Couldn't set sample rate\n");
            exit(0);
        }
        SoapySDRKwargs args = {0};
        r = SoapySDRDevice_setFrequency(device, SOAPY_SDR_RX, 0, (double)868412500, &args);
        if(r != 0){
            printf("Couldn't set frequency\n");
            exit(0);
        }
        //args = {0};
        r = SoapySDRDevice_setGain(device, SOAPY_SDR_RX, 0, 34);
        if(r != 0){
            printf("Couldn't set gain\n");
            exit(0);
        }

		args = {0};
        //SoapySDRDevice_setupStream(device, &stream, direction, SOAPY_SDR_CU8, 0, 2, &args);

        if(SoapySDRDevice_activateStream(device, stream, 0, 0, 0) != 0){
            printf("Couldn't activate stream\n");
            exit(0);
        }
        suppress_stdout_stop(tmp_stdout);

    return 0;
}

int Radio::getStream(float **output, int &outSize)
{
    void *buffs[] = {buffer};

    //std::vector<liquid_float_complex> in_data;
    //std::vector<fdata> in_data;

    //int ret;
        outbuff = (float*)malloc(out_block_size*sizeof(float)*2);
        int n_stream_read = 0;
        n_read = 0;
        int n_stream_written = 0;
        while(n_read < out_block_size){
            n_stream_read = SoapySDRDevice_readStream(device, stream, buffs, out_block_size, &flags, &timeNs, timeoutNs);
            if(n_stream_read == SOAPY_SDR_OVERFLOW){
                outSize = 0;
                fprintf(stderr, "O");
                fflush(stderr);
                continue;
            } else {
                n_read += n_stream_read;
                for(int i = 0; i < n_stream_read; i++){
                    outbuff[i+n_stream_written] = buffer[i];
                }
                n_stream_written = n_stream_read;
            }

        }
                fwrite(buffer, sizeof(float), n_read, stdout);
                fflush(stdout);
        fprintf(stderr, "%d\n", n_stream_read);
        //usleep(50);
        //device->readStream(stream, buffs.data(), numElems, flags, timeNs);
        //int n_stream_read = device->readStream(stream, buffs, numElems, flags, timeNs);
        /*outbuff = (float*)malloc(n_stream_read*sizeof(float)*2);
        n_read = n_stream_read*2;
        for(int i = 0; i < n_read; ++i){
            buf8[i] = ( (int16_t)buffer[i] / 32767.0f * 128.0f + 127.4f);
        }
        //*output = (float*)malloc(n_stream_read*sizeof(float) * 2);
        //printf("%f\n", n_stream_read);
        outSize = n_stream_read;*/
    /*try {
    } catch (const std::exception &ex) {
        std::cerr << "Error probing device: " << ex.what() << std::endl;
        return 1;
    }*/
    //state collected in this loop
    //unsigned int overflows(0);
    //unsigned int underflows(0);
    //unsigned long long totalSamples(0);
    //int spinIndex(0);
    /*for(int i=0;i<outSize*2;i++){
        outbuff[i]=((float)buf8[i])/(UCHAR_MAX/2.0)-1.0; //@convert_u8_f
    }
    *output = outbuff;*/
    return 0;

}

void Radio::uninit_radio()
{
    ///device->deactivateStream(stream);
    SoapySDRDevice_deactivateStream(device, stream, 0, 0);
    //SoapySDR::Device::unmake(device);
    /*hackrf_close(hackrf);
    hackrf_exit();*/
}

int Radio::verbose_device_search(char *s, SoapySDRDevice **devOut, SoapySDRStream **streamOut, const char *format){
    size_t device_count = 0;
	//size_t i = 0;
	int device, offset;
	char *s2;
	char vendor[256], product[256], serial[256];
	SoapySDRDevice *dev = NULL;

	SoapySDRKwargs stream_args = {0};

	dev = SoapySDRDevice_makeStrArgs(s);
	if (!dev) {
		fprintf(stderr, "SoapySDRDevice_make failed\n");
		return -1;
	}

	//show_device_info(device);

	if (SoapySDRDevice_setupStream(dev, streamOut, SOAPY_SDR_RX, format, NULL, 0, &stream_args) != 0) {
		fprintf(stderr, "SoapySDRDevice_setupStream failed\n");
		return -3;
	}

	*devOut = dev;
	return 0;
}

int Radio::suppress_stdout_start(void) {
	int tmp_stdout = dup(STDOUT_FILENO);
	if (dup2(STDERR_FILENO, STDOUT_FILENO) != STDOUT_FILENO) {
		perror("dup2 start");
	}

	return tmp_stdout;
}

void Radio::suppress_stdout_stop(int tmp_stdout) {
	fflush(stdout);
	if (dup2(tmp_stdout, STDOUT_FILENO) != STDOUT_FILENO) {
		perror("dup2 stop");
	}
}

```

guruofquality, 2017-11-10:
It may help to look at some examples like rxtools: https://github.com/rxseger/rx_tools/tree/master/src

* Whats the sample type format? The buffers are casted as floats, but I dont see the format being complex float in the code?
* Also, Keep in mind that the samples are complex, so if you are casting to floats, the number of floats in the buffer = the number samples received times 2.

navysurfer, 2017-11-10:
Thanks for the reply.  I actually got this far from taking apart the rx_tools source.  The stream format is CS16.  I'm not trying to change it, I'll figure all that out later on.
  Thanks for pointing out the wrong casts too.  I'll take a look at that in a few hours and report back.

navysurfer, 2017-11-11:
Okay, several more hours into it, I decided to compile the rx_tools and run that, and it was giving me the same problems.  I get about 1/4-1/2 sec static, long pause, then static again.  It sounds just like oversampling.  

I thought maybe it was a library problem, or a device conflict, but CubicSDR compiles and runs fine, but that uses the C++ library and not the C.  I started with the c++ library, as you can probably tell by the commented code in my first post. After a week of reading through cubic's source, I decided to look for something simpler to start with.  
 
I still went in and fixed the bad casting you pointed out earlier.  I had some code in there previously to convert the array to a float array and forgot to change it back when I removed the code.  I went through and cleaned up my code the best I can to try to identify the problem, but it's behaving exactly the same way as rx_tools.  Thanks again for the help.  Sorry for the long post.

Header

```
#ifndef RADIO_H
#define RADIO_H

#include <iostream>
#include <unistd.h>

extern "C" {
    #include <SoapySDR/Device.h>
    #include <SoapySDR/Formats.h>
}

#define DEFAULT_BUF_LENGTH		(16 * 16384)

class Radio
{
    public:
        Radio();
        int initialize_radio();
        int getStream(float **output, int &outSize);
        void uninit_radio();
        int16_t *buffer;
        SoapySDRDevice *device;
        SoapySDRStream *stream;

    private:
        uint32_t out_block_size = DEFAULT_BUF_LENGTH;
        int verbose_device_search(char *s, SoapySDRDevice **devOut, SoapySDRStream **streamOut, const char *format);
        int suppress_stdout_start(void);
        void suppress_stdout_stop(int tmp_stdout);
        int flags = 0;
        long long timeNs = 0;
        int n_read = 0;
        long timeoutNs = 10000000;
};

#endif // RADIO_H
```

Source

```
#include "Radio.h"

Radio::Radio()
{
    //ctor
}

int Radio::initialize_radio(){
    int tmp_stdout = suppress_stdout_start();
	buffer = (int16_t*)malloc(out_block_size * SoapySDR_formatToSize(SOAPY_SDR_CS16));
    int r = verbose_device_search("", &device, &stream, SOAPY_SDR_CS16);
    if(r != 0){
        fprintf(stderr, "Couldn't find SDR Device\n");
        exit(0);
    }
    double sampleRate = 2400000.0;
    fprintf(stderr, "Reading at Sample Rate: %d\n", (long)sampleRate);
    r = SoapySDRDevice_setSampleRate(device, SOAPY_SDR_RX, 0, (double)sampleRate);
    if(r != 0){
        fprintf(stderr, "Couldn't set sample rate\n");
        exit(0);
    }
    SoapySDRKwargs args = {0};
    r = SoapySDRDevice_setFrequency(device, SOAPY_SDR_RX, 0, (double)868412500, &args);
    if(r != 0){
        fprintf(stderr, "Couldn't set frequency\n");
        exit(0);
    }
    r = SoapySDRDevice_setGain(device, SOAPY_SDR_RX, 0, 34);
    if(r != 0){
        fprintf(stderr, "Couldn't set gain\n");
        exit(0);
    }

    if(SoapySDRDevice_activateStream(device, stream, 0, 0, 0) != 0){
        fprintf(stderr, "Couldn't activate stream\n");
        exit(0);
    }
    suppress_stdout_stop(tmp_stdout);
    return 0;
}

int Radio::getStream(float **output, int &outSize)
{
    void *buffs[] = {buffer};
    int n_stream_read = 0;
    n_stream_read = SoapySDRDevice_readStream(device, stream, buffs, out_block_size, &flags, &timeNs, timeoutNs);
    n_read = n_stream_read*2;
    fwrite(buffer, sizeof(int16_t), n_read, stdout);
    fflush(stdout);
    for(int i = 0; i < n_read; i++){
        if(buffer[i] > 0 || buffer[i] < 0){
            fprintf(stderr, "%d\n", buffer[i]);
        }
    }
    return 0;
}

void Radio::uninit_radio()
{
    SoapySDRDevice_deactivateStream(device, stream, 0, 0);
}

int Radio::verbose_device_search(char *s, SoapySDRDevice **devOut, SoapySDRStream **streamOut, const char *format){
    size_t device_count = 0;
	int device, offset;
	char *s2;
	char vendor[256], product[256], serial[256];
	SoapySDRDevice *dev = NULL;
	SoapySDRKwargs stream_args = {0};
	dev = SoapySDRDevice_makeStrArgs(s);
	if (!dev) {
		fprintf(stderr, "SoapySDRDevice_make failed\n");
		return -1;
	}
	if (SoapySDRDevice_setupStream(dev, streamOut, SOAPY_SDR_RX, format, NULL, 0, &stream_args) != 0) {
		fprintf(stderr, "SoapySDRDevice_setupStream failed\n");
		return -3;
	}
	*devOut = dev;
	return 0;
}

int Radio::suppress_stdout_start(void) {
	int tmp_stdout = dup(STDOUT_FILENO);
	if (dup2(STDERR_FILENO, STDOUT_FILENO) != STDOUT_FILENO) {
		perror("dup2 start");
	}
	return tmp_stdout;
}

void Radio::suppress_stdout_stop(int tmp_stdout) {
	fflush(stdout);
	if (dup2(tmp_stdout, STDOUT_FILENO) != STDOUT_FILENO) {
		perror("dup2 stop");
	}
}
```

navysurfer, 2017-11-11:
Okay, I figured it out.  It turns out that both `SoapySDRDevice_makeStrArgs("")`and `SoapySDRDevice_make(&args_with_empty_strings)` return a device pointer.  So after that point, it still let me set everything on it like there was no issue.  I'll post the corrected code if you'd like to have it here.  Thanks for the help.

guruofquality, 2017-12-10:
ty!

closed 2017-12-10 by guruofquality