pothosware/SoapySDR #148: SoapySDR issue with LimeSDR-mini and gr-osmosdr

kantooon, 2017-12-02:
Hello,

I am adding support for the LimeSDR-mini in the QRadioLink transceiver ( https://github.com/kantooon/qradiolink ) and I encountered an issue which I think comes from SoapySDR. Upon starting a flowgraph (which is a basic Gnuradio flowgraph using the gr-osmosdr sink), if I transmit audio immediately to the flowgraph the stream seems to stall and nothing gets transmitted. Introducing a 100 msec delay between the two operations seems to solve the problem in most cases, however the latency makes this a no go for me.
Need to mention that I don't have the same issue with UHD and USRP.

Thanks,
Adrian

guruofquality, 2017-12-02:
Can you share any details about what "transmit audio immediately to the flowgraph" implies? Is it a continuous stream of samples or a small burst of samples perhaps? That would help determine if we are looking at some kind of flushing issue in the LimeSDR or maybe an delay issue from large buffers in the LimeSDR driver.

If it helps, the gr-osmosdr soapy stuff and soapylms stuff is very thin, its really just adapting API calls, it doesnt itself buffer or delay anything.

> Need to mention that I don't have the same issue with UHD and USRP.

Might be worth trying the same with SoapyUHD installed to isolate things to just gr-osmosdr's soapy support vs gr-osmosdr's uhd support or gr-uhd?

Since you are using gnuradio, are there any use of stream tags that need to be supported (for example) for you application?

kantooon, 2017-12-02:
Hi Josh,

The sequence is as follows: first I call the start() method of the flowgraph, and then I push a vector of chars representing encoded digital audio into a vector source to be ultimately transmitted as a BPSK/QPSK/FSK signal. The transmission is continuous, lasting for as long as the user keeps the PTT activated. 

With LimeSDR, if I don't introduce a rather large delay between these operations, nothing gets transmitted. There are (currently) no stream tags needed although SOB/EOB might be required in the future.

I will try the SoapyUHD variant, although gr-osmosdr does not seem to have any issues apart from this one.

Thanks,
Adrian

guruofquality, 2017-12-03:
I have a suspicion that the 100ms is because the first call to start() is performing a self calibration in SoapyLMS. Although I'm still not sure why this would make sense since start() is blocking, unless perhaps the two operations can run in parallel with multiple threads. So if that makes sense, perhaps SoapyLMS needs to hold off on the sample transmission (writeStream) while start() (activate) completes.

kantooon, 2017-12-03:
Ok, thanks, that sounds plausible. The issue is that nothing gets transmitted at all anymore after that unless first destroying the osmosdr sink and re-creating it. Is there anything that can be done about this in SoapyLMS, to avoid an arbitrary length sleep before sending data to the device?

kantooon, 2017-12-14:
I don't think there is anything that can be done about this in SoapySDR or QRadiolink , so I'll close this issue.

closed 2017-12-14 by kantooon