pothosware/SoapySDR #155: SoapySDRUtil: symbol lookup error

MStrecke, 2018-01-10:
I want to use an SDRPlay RSP 2 (USB ID 1df7:3010) with gqrx under Lubuntu 17.10.

I've started with a fresh install of Lubuntu 17.10 and installed gqrx as described [here](http://gqrx.dk/download/install-ubuntu).

I've then followed the steps outlined in http://www.sdrplay.com/docs/SDRplay_non_Windows_Flow.pdf

I've downloaded and run: SDRplay_RSP_API-Linux-2.11.1.run
Then I compiled SoapySDR and SoapySDRPlay as described in the PDF.

There weren't any obvious error messages during compilation (see attached log).
However the installation tests fail with the following error message:

```
$ SoapySDRUtil –-info
SoapySDRUtil: symbol lookup error: SoapySDRUtil: undefined symbol: _ZN8SoapySDR17ConverterRegistry26listAvailableSourceFormatsB5cxx11Ev
```

Any hints?

[compilelog.txt](https://github.com/pothosware/SoapySDR/files/1618941/compilelog.txt)

guruofquality, 2018-01-10:
since `SoapySDR::ConverterRegistry::listAvailableSourceFormats` is a recent addition, looks like SoapySDRUtil and all was compiled recently on the master branch, but its using an old libSoapySDR? Can you see if there are any old libSoapySDR* files hanging around in /usr/lib or /usr/local/lib? Are you mixing release packages with source installs for the same software, etc?

MStrecke, 2018-01-11:
As mentioned above, this was done on a freshly installed system. The only files came from "SDRplay_RSP_API-Linux-2.11.1.run" (downloaded a few days ago from the Website) and the files cloned from github  as described in the Non-Windows-Flow.pdf.

Due to another showstopper in gqrx, I wiped the disk and freshly installed Xubuntu 16.04.3 LTS on that system.

Same procedure as before (installing gqrx from the repos, installing the API file, compiling according to the PDF). Again, no obvious errors. This time, I get another symbol error:

```
$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-myriadrf~xenial
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/i386-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/i386-linux-gnu/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
SoapySDRUtil: symbol lookup error: SoapySDRUtil: undefined symbol: _ZN8SoapySDR16getModuleVersionERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
```
Progress?

Thank you for your help.

guruofquality, 2018-01-11:
So it looks like you are installing two different versions of SoapySDR at the same time, one from the package manager (apt-get) as seen by the prints above (`Lib Version: v0.6.1-myriadrf~xenial`), and another version from source build. I'm sure that's the cause of the problem.

From the verbose above it looks like soapysdrplay compiled against the source based install in /usr/local, but its being loaded by the binary install in /usr/. Can you make sure only one version of SoapySDR is installed and recompile SoapySDRPlay against that one version?

guruofquality, 2018-01-12:
Looks like this is basically a duplicate issue, the errors change because the development branch has moved on from the previous release, but the cause is basically the same: https://github.com/pothosware/SoapyRemote/issues/31

Essentially the gist of it is:

* use apt repo for library and some modules, build sources modules that are not in the repo
  * (this also means installing libsoapysdr-dev to get the matching development files)
* or use all source builds for the main library and modules and remove anything from apt repo

closed 2018-01-12 by guruofquality

MStrecke, 2018-01-12:
Yes, you're right. I didn't realized that.

Thanks