pothosware/SoapySDR #156: Spyserver support

nmaster2042, 2018-01-17:
Hi,

The airspy team develops a server to use airspy, airspy mini, and airspy hf+ remotely over ip just like rtl_tcp does for RTLSDRs.

Some servers are publicly avaliable (see airspy.comweb site).

It would be a nice addition to have a SoapySpyserver to handle theses servers.

There is an opensource example of spyserver client implementation here:

https://github.com/opensatelliteproject/xritdemod/blob/master/demodulator/src/SpyServerFrontend.cpp

guruofquality, 2018-01-17:
Not the first time it came up, but FWIW I coudn't find the spec, so I'm glad there is some open source code to show the format if thats what it takes.

Also, is the idea to make a server that emulates the airspy network protocol so any SoapySDR device can appear as a remote airspy device?

Or is the idea to make a client that can talk to an airspy server and wrap that into a SoapySDR support module?

Or both? :-)

nmaster2042, 2018-01-17:
Hi guruofquality,

When I first heard of spyserver, I though it was closed, the source of this software isn't published.

But, I red the airspy team let people implement the protocol used by spyserver in order to add spyserver support on open source software clients.

The idea was to create a SoapySpyserver so that softwares like CubicSDR or GQRX can connect and use spyservers on a LAN or on the net.

For the idea of creating a server using spyserver protocol, I don't know if airspy team will allow it, as it's closed source.

Moreover, isn't SoapyRemote doing the same thing ?

Not exactly, because spyserver seems to use compression, to minimize bandwith usage.
It can send full iq or small part iq that is actually needed for demodulation on the client and it is not multi user.

But why not extend SoapyRemote, adding such optimization ideas, without using the spyserver protocol ? so that it becomes doable to share a soapy sdr device on the net ?

nmaster2042, 2018-07-01:
Hi,

There is a new gr-osmosdr fork that provide spyserver support:
https://github.com/racerxdl/gr-osmosdr

Tried it with gqrx, it's working fine.

Perhaps a good start for a SoapySpyserver implementation ?

ckarrie, 2021-01-15:
Would love to see it in soapySDR

ncorgan, 2022-03-14:
Closed with the release of https://github.com/pothosware/SoapySpyServer

closed 2022-03-14 by ncorgan