pothosware/SoapySDR #159: Soapy spamming hackrf_exit() failed - no hackrf connected

alexwhittemore, 2018-02-21:
I'm very new to Soapy and the LimeSDR I'm trying to fly, so forgive my ignorance on basically every piece of this toolchain. I've got an issue while I'm trying to figure everything out that Soapy seems to be spamming things about hackrf, mainly to the Pothos output window when I'm trying to set up a flow. Mostly, it's annoying visual clutter, but I'm worried it means something's wrong under the hood that's going to bite me. I've never owned or connected a HackRF to this computer, so I'm not sure what's causing this.

For example, in PowerShell:
```
PS C:\Users\alexw> SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.003.000-0-gef157678

[1m[31m[ERROR] hackrf_exit() failed -- some hackrf is sill in use[0m
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 3
  driver = audio
  label = Headset Microphone (Realtek High Definition Audio(SST))

Found device 2
  default_input = False
  default_output = False
  device_id = 4
  driver = audio
  label = Microphone Array (Realtek High Definition Audio(SST))

Found device 3
  driver = lime
  index = 0
  label = USB 3.0 (LimeSDR-USB) [USB] 9060B00463019
  media = USB
  module = FX3
  name = USB 3.0 (LimeSDR-USB)
  serial = 0009060B00463019
```

In Pothos Flow, the minute I add a SoapySDR Source, in the Message window:
```
[01:01:38.627000] SoapySDR: hackrf_exit() failed -- some hackrf is sill in use
[01:01:44.473000] SoapySDR: hackrf_exit() failed -- some hackrf is sill in use
```
This repeats as long as the block still exists.

Is something wrong? Do I just deal with this?

guruofquality, 2018-02-21:
Its a recent issue, no resolution yet, but here is the issue: https://github.com/pothosware/SoapyHackRF/issues/20 but In the meantime, you can delete program files/PothosSDR/lib/SoapySDR/modules0.6/hackrfsupport.dll and that should go away https://groups.google.com/forum/#!topic/pothos-users/IqTWedSMwjE

closed 2018-02-21 by guruofquality

alexwhittemore, 2018-02-21:
Ahh, thank you for pointing me to the right spot and to a workaround!