pothosware/SoapySDR #160: SoapySDR - no modules found and no factories found

ramusters, 2018-03-19:
I am trying to install an Airspy mini onto a raspberry Pi 3 using SoapySDR.

After following the build guide and running SoapySDRUtil --info I get: No modules found! and No factories found!

Below the entire output:

After doing 
mkdir build 
cd build:

pi@raspberrypi:~/SoapySDR/build $ cmake ../ -DCMAKE_BUILD_TYPE=Release
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
-- PYTHONINTERP_FOUND: TRUE - 3.5.3
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python3/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 3.5.3
-- PYTHON_INCLUDE_DIRS: /usr/include/python3.5m
-- PYTHON_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpython3.5m.so
-- CMAKE_SWIG_FLAGS=-c++;-threads;-DSIZE_T_IS_UNSIGNED_INT
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.10
-- Could NOT find Python3InterpDbg (missing:  PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
-- PYTHON3_INSTALL_DIR: ${prefix}//usr/lib/python3.5
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.5m
-- PYTHON3_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpython3.5m.so
-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library , runtime library v0.7.0-g824d83e8
 * Tests , library unit tests
 * Docs , doxygen documentation
 * Python , python bindings v3.5.3

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 
 * Python3 , python3 bindings

-- SoapySDR version: v0.7.0-g824d83e8
-- ABI/so version: v0.7
-- Install prefix: /usr/local/lib/python3.5
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/SoapySDR/build
pi@raspberrypi:~/SoapySDR/build $ make -j4
Scanning dependencies of target SoapySDR
[  2%] Built target docs
[  5%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Modules.cpp.o
[  8%] Linking CXX shared library libSoapySDR.so
[ 63%] Built target SoapySDR
[ 66%] Linking CXX executable TestTimeConversion
[ 69%] Linking CXX executable TestKwargsMarkup
[ 72%] Linking CXX executable TestFormatParser
[ 75%] Linking CXX executable SoapySDRUtil
[ 77%] Built target TestTimeConversion
[ 80%] Built target TestFormatParser
[ 83%] Built target TestKwargsMarkup
[ 91%] Built target SoapySDRUtil
[ 94%] Linking CXX shared module _SoapySDR.so
[100%] Built target _SoapySDR
pi@raspberrypi:~/SoapySDR/build $ sudo make install
[ 61%] Built target SoapySDR
[ 72%] Built target SoapySDRUtil
[ 77%] Built target TestTimeConversion
[ 83%] Built target TestFormatParser
[ 88%] Built target TestKwargsMarkup
[ 91%] Built target docs
[100%] Built target _SoapySDR
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/python3.5/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Installing: /usr/local/lib/python3.5/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Installing: /usr/local/lib/python3.5/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Installing: /usr/local/lib/python3.5/include/SoapySDR
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Config.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Config.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Version.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Logger.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Modules.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/ConverterRegistry.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Registry.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Types.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Time.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Errors.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Logger.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Formats.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Formats.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Constants.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Version.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Time.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Modules.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Device.h
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Types.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Device.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/ConverterPrimatives.hpp
-- Installing: /usr/local/lib/python3.5/include/SoapySDR/Errors.hpp
-- Installing: /usr/local/lib/python3.5/lib64/libSoapySDR.so.0.7.0
-- Installing: /usr/local/lib/python3.5/lib64/libSoapySDR.so.0.7
-- Installing: /usr/local/lib/python3.5/lib64/libSoapySDR.so
-- Installing: /usr/local/lib/python3.5/lib64/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/lib/python3.5/bin/SoapySDRUtil
-- Set runtime path of "/usr/local/lib/python3.5/bin/SoapySDRUtil" to ""
-- Installing: /usr/local/lib/python3.5/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/lib/python3.5/lib/python3/dist-packages/_SoapySDR.so
-- Set runtime path of "/usr/local/lib/python3.5/lib/python3/dist-packages/_SoapySDR.so" to ""
-- Installing: /usr/local/lib/python3.5/lib/python3/dist-packages/SoapySDR.py
pi@raspberrypi:~/SoapySDR/build $ sudo ldconfig
pi@raspberrypi:~/SoapySDR/build $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.0-g824d83e8
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.7
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

guruofquality, 2018-03-19:
Looks good to me. If you are using airspy, go ahead and install its hardware support module: https://github.com/pothosware/SoapyAirspy/wiki

ramusters, 2018-03-19:
Ok! I reckoned no modules found meant something went wrong. I have SoapyAirspy installed and after typing SoapySDRUtil --find I get 
No devices Found

Below the entire output:
pi@raspberrypi:~ $ git clone https://github.com/pothosware/SoapyAirspy.git
fatal: destination path 'SoapyAirspy' already exists and is not an empty directory.
pi@raspberrypi:~ $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found!
pi@raspberrypi:~ $

guruofquality, 2018-03-19:
Can you share the output of the build (cmake, make, make install) for SoapyAirspy, just like you did for SoapySDR?

If you do have the hardware support properly installed, its best to check lsusb if the device is found, and then check if you can find the device with sudo. If only root works, you need the udev rules for that particular device.

closed 2018-04-09 by guruofquality

raymiec, 2022-05-19:
@ramusters  trying to contact you, can you DM me